xodio/xod #725: As a user I want everything to have tooltips

daphee, 2017-08-13:
## Feature/Story Template 

### Rationale

As of now I couldn't find tooltips anywher. They would help new and experienced users alike by sparing them a trip to the documentation.

### Acceptance criteria

- [ ] Hover over any input field/button/... to get a description of what it does.
- [ ] Inputs and outputs of node should have a brief description as well

### How to implement

Tooltips can be added to HTML by using ``title`` attribute. Node inputs/outputs can get a tooltip by using a ``<title></title>`` child in the SVG.

nkrkv, 2017-08-14:
Welcome!

The criteria are too wide so we can’t arrange the issue for development. Making a tooltip for *any* field and button doesn’t make sense. E.g. I see no point in providing a title for the New Patch Name field in the New Patch dialog as it already accompanied by a label. The same for OK button in that dialog.

Could you narrow down the context? E.g. list concrete UI elements which lacks tooltip and for which a title makes much sense. Don’t be afraid to make the issue small. It’s much better to have many small and precise issues that few very broad.

What about tooltips for inputs and outputs, they are covered by #34

daphee, 2017-08-14:
I would propose tooltips for the following:

- *Label a node* (textbox with green flag) and *View documentation* button. Pretty self explanatory but I personally still like to know exactly what a button does before pressing it. Even more so a button that does an action (opening a new browser window) outside of the application. So preferably the tooltip would for the *View documentation* would read something like "Open documentation for this node **in your browser**".  (See [here](https://i.imgur.com/Ds8IW5v.png))

- *Add node* and *View node documentation* button ([here](https://i.imgur.com/ZqO325b.png)) 

- *Delete patch* and *Rename patch* in "My Patches"

The mockup you posted in #34 looks great. I have some comments but I'll add them there directly.

nkrkv, 2017-08-14:
Thank you for the details. I expanded the issue body and moved it to the Backlog.

commit a2974056 (2017-08-29): feat(xod-client): add tooltips for unlabeled fields/buttons in ProjectBrowser and Inspector

Closes #725
PR #752 (2017-08-29): feat(xod-client): add tooltips for unlabeled fields/buttons in ProjectBrowser and Inspector
Closes #725
closed 2017-08-29 by evgenykochetkov