xodio/xod #734: Projects should survive integrity break on load

nkrkv, 2017-08-17:
Now if we’d try to open a project which is fine from JSON and data shape point of view, but which have integrity issues such as bad patch path references and bad link references, we’ll fail. IDE hangs. That’s very frustrating. We need a mechanism which allows to go on.

We can still draw broken nodes and links with some special style (dashed red). There should be enough information for that: patch path is known, the fact the pin was there is known too. The info about the label is lost, but that’s OK.

### Test scenario

1. Make a blink patch: clock → flip-flop → led
2. Save it and close IDE
3. Delete the `flip-flop` from `lib/xod/core`
4. Launch IDE and try to open that patch

### Acceptance criteria

- [ ] The patch gets opened successfully. The `flip-flop` and its links are drawn with the broken style
- [ ] I can select the broken node, drag it around, select, delete
- [ ] I can select the broken links and delete them
- [ ] I can save/load a project with broken stuff and get the same result
- [ ] When I try to transpile I get a clear error message like “Patch @/main contains broken references. Fix or delete them to continue.”
- [ ] I can close IDE, restore the `flip-flop`, open the project again and get the state I’d get if the `flip-flop` was never deleted
- [ ] The same round trip is possible for another scenario when I only delete the `TGL` pin in `flip-flop`. In that case, only a single link would be considered broken.

evgenykochetkov, 2017-08-17:
How far should we propagate info about failed integrity check?
For example, if we have `@/broken-patch` which uses an unknown node, should the  `@/broken-patch` node in `@/main` be drawn red? What about the Project Browser?

nkrkv, 2017-08-17:
> For example, if we have `@/broken-patch` which uses an unknown/broken node, should the  `@/broken-patch` node in `@/main` be drawn red?

Doesn’t matter much IMO if the next point would be done

> What about the Project Browser?

Lets mark patches with broken content and projects with broken patches

PR #815 (2017-10-19): Add surviving of the project
It closes #734 .
closed 2017-10-19 by brusherru