xodio/xod #755: Tab ordering in the Inspector is broken

nkrkv, 2017-08-30:
### Steps to reproduce

- Place a `text-lcd-16x2` node
- Click `RS` value input in the Inspector to focus it
- Enter a value (e.g. `8`) and press `TAB`

### Expected behavior

- The cursor moves to the next pin input (`EN`), its text is selected
- I can quickly enter a value → `TAB` → enter value → `TAB` → … → enter value → `ENTER` to bind all pin values.

### Actual behavior

- The focus is lost
- Pressing `TAB` more times brings the focus back to the `RS` input
- Pressing `TAB` again looses focus
- Pressing `TAB` one more time focuses the `EN`’s input
- I have to press `TAB` twice to move to the next input

evgenykochetkov, 2017-10-30:
Note: this will most likely be quite tedious, because our widgets system has grown into a bit of a mess. For example, the problem causing this specific issue is that almost the whole inspector rerenders after a single value change(see https://github.com/xodio/xod/blob/master/packages/xod-client/src/editor/components/WidgetsGroup.jsx#L11 etc)
All of those problems should be gone after #766

nkrkv, 2017-10-30:
So now #766 has a clear rationale to be done. I’ll set it as a prerequisite.

PR #1681 (2019-02-13): Refactor inspector widgets
Closes #766
Closes #755 (`tab` now works properly)

and also:
* after committing values with Enter key, the whole contents of the input is selected
* incrementing with up/down arrows works again on Number widgets (default step is 0.1, with shift key it's 1)
closed 2019-02-13 by evgenykochetkov