xodio/xod #786: "uploading project" modal gets stuck if serial port in use or unavailable

awgrover, 2017-09-25:
deskstop xod 0.13.0 from .deb install
linux mint 17.1 (ubuntu 14.04 equiv)
Linux 3.13.0-37-generic #64-Ubuntu SMP Mon Sep 22 21:28:38 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

## Bug Report

### Steps to reproduce

- Open welcome-to-xod
- arduino is attached
- arduino IDE is open and serial console is open
- "uploading project" modal opens. "Toolchain is installed. Uploading..."
- alert dialog-window appears:
`A JavaScript error occurred in the main process`
`Uncaught Exception:
Error: Error: Device or resource busy, cannot open /dev/ttyACM0`
- click "ok"

### Expected behavior

- upload-modal should close
- or, steal port from arduino ide

### Actual behavior
- modal stays, canvas is not usable. have to use something from the "Deploy" menu to make it go away

- similar behavior if
-- arduino is not attached
-- then upload

### Console output

developers tools, console shows no output

nkrkv, 2017-09-25:
Welcome! Thanks for the report.

I’ve scheduled the issue in our backlog.

nkrkv, 2017-10-04:
Fixed by #797

closed 2017-10-04 by brusherru