xodio/xod #787: as a native node developer, I want a quick "reload, compile & upload project"

awgrover, 2017-09-27:
### Rationale

I'm developing some native nodes, and repeatedly go through the cycle:
* Edit my lib's .xodp and .cpp
* reload the project in XOD-desktop (two steps)
* upload (no hotkey)
* get gcc errors, or check the arduino serial-console
* if gcc error, get the code ("show code for arduino"), and hunt for the line, and try to guess what gcc means :)

That's tedious and slow.

### Prerequisites

No obvious prerequisites

related #459 #332 

### Acceptance criteria

I can test my native-node libs by pushing one button, or using a command-line (command line preferred), and get the results without having to open panes/windows/modals (preferable via the terminal and file-system), so the edit/test cycle is faster/immediate.  I have to setup a project that uses my native-node lib.

- [ ] push one button, or using a command-line (command line preferred)
- [ ] it does the equivalent of "reload", using this project, but with the updated /lib
- [ ] and compiles
- [ ] and outputs the validation/gcc error (console preferred)
- [ ] keep the generated code around (equivalent of "show arduino code") as a file
- [ ] enabling the various debug features (trace, and other suggested features like possibly #33, #780)
- [ ] uploads to arduino (if it gets this far)
- [ ] auto magically opening the equivalent of the serial-monitor (this would be real nice, but could be "later")

### Out of scope

### How to implement

Via the command line would be preferred (allows integration with editors, etc.)

Via a hot-key should also be supported.

### How to test

1.
- [ ] Build a "good" test project, with a "good" test lib (native node).
- [ ] as a 'test', do the "one-button" "testit" thing (again, command-line preferred)
- [ ] gcc step should not fail
- [ ] generated code is kept

2. 
- [ ] Build a "bad" test project, with a "bad" test lib (native node), with a known c++ error.
- [ ] as a 'test', do the "one-button" "testit" thing (again, command-line preferred)
- [ ] gcc step should fail with known error
- [ ] gcc error output is kept
- [ ] generated code is kept

3. 
- [ ] Build a "bad" test project, with a "bad" test lib (native node), with a known error (not in c++, e.g. bad xodp etc.).
- [ ] as a 'test', do the "one-button" "testit" thing (again, command-line preferred)
- [ ] pre-gcc step (validation?) should fail with known error
- [ ] error output is kept
- [ ] generated code is kept (if any)

nkrkv, 2017-09-27:
I see your pain. Thank you for filing the issue. However, as developers, we want to keep the issue list as minimal and as precise as possible. In other words, absolutely all open issues should have (a) an obvious notion of “done”, (b) they are agreed to be included in the product.

In this particular issue I see the problem, but in the same time, I see several absolutely different possible solutions (upload with CLI, embedded C++ editor, manual project reload, hot reload, re-deploy menu item) implementation of which could be split significantly in time. For the reasons mentioned, there should be a discussion prior to the feature issue creation. [XOD forum](https://forum.xod.io) is a good place to decide possible problems and their solutions as it has better visibility for xoders. I should definitely make this clear in the `ISSUE_TEMPLATE.md` and `CONTRIBUTING.md`.

For now, if you don’t mind, could we move the discussion to the forum and close the issue as is? As a result of the discussion, we’ll either reopen this issue or (likely) split it and file several smaller and specific issues.

awgrover, 2017-09-27:
no problem. I'll make a forum topic

closed 2017-09-27 by awgrover