xodio/xod #888: Can’t bind values to constants

nkrkv, 2017-11-15:
### Steps to reproduce

1. Place a `xod/core/constant-number` node
2. Try to bind a value

### Expected behavior

- I can enter the value on `VAL` output via Inspector

### Actual behavior

- The field is disabled saying “defined by input”

evgenykochetkov, 2017-11-16:
I can't reproduce it(both in the released version and most recent master),
and we already have a check for this case:
https://github.com/xodio/xod/blob/6d0266f836074f88f4f359bd403b2a08763d7b67/packages/xod-project/src/patch.js#L268-L280

But it **does** reproduce when `xod/core` as opened as a project, and constant nodes have type `@/constant-number ` instead of `xod/core/constant-number`.

nkrkv, 2017-11-16:
@evgenykochetkov Ah, I see. I think it’s a ~bug~ flaw anyway but given it appears only in a rudimental case, it is not a regression, nor urgent. Thank you.

brusherru, 2017-11-16:
@nkrkv I reproduced this bug. And I think this is urgent and regression 😬 
It occurred when I test wave nodes and tried to set `T` for all nodes by bind some value to the constant-number. I tested them in the separate project (not inside `xod/core`).

nkrkv, 2017-11-17:
:open_mouth: I’m at 6d0266f836074f88f4f359bd403b2a08763d7b67 and can’t reproduce the bug in `welcome-to-xod`. @brusherru Can you spot the exact step-by-step scenario? Commit SHA you’re on will also help.

evgenykochetkov, 2017-11-17:
Here is a minimal example to reproduce the issue @brusherru is talking about: https://gist.github.com/evgenykochetkov/ce828868a657f7bfd529631a96f88cac

But that is another issue, not the one described here.