xodio/xod #896: [Feature request] Scrolling without a mouse

sergeysi779, 2017-11-21:
As of now patch scrolling is only possible by means of mouse middle button (please correct me if I'm wrong). It would be nice to have scrolling functionality with a touchpad or a touchscreen.

nkrkv, 2017-11-21:
Hello!

Please, post feature requests on [XOD forum](https://forum.xod.io). GitHub issues are things that *should* be done and they’re not the best place to discuss. Feature issues are created as a result of discussions on the forum.

I agree we need to improve scrolling but at a first sight, I see three different features/opportunities: scrollbars, arrow keys, scroll wheel (double finger touchpad gesture). So a preliminary discussion is required.

Until that, I have to close the issue.

closed 2017-11-21 by nkrkv

evgenykochetkov, 2017-11-21:
@sergeysi779 Patch scrolling is also possible while holding spacebar and dragging the mouse.