xodio/xod #964: Building XOD Desktop IDE

puttley, 2017-12-18:
I am having difficulty building the XOD desktop IDE from the cloned source files. I'm pretty sure that I have all of the necessary modules installed correctly, but build continues to fail. 
I have attached the lerna and npm debug log files. Hopefully someone has experienced this and can provide guidance. Thank you.
[npm-debug.log](https://github.com/xodio/xod/files/1569383/npm-debug.log)
[lerna-debug.log](https://github.com/xodio/xod/files/1569384/lerna-debug.log)

evgenykochetkov, 2017-12-18:
Hi!
Please try using node 8(or higher) and latest yarn and report if that helps.

puttley, 2017-12-19:
Thanks for the suggestion. I have followed instructions for updating node and  yarn, but am not sure that it was successful. I have found that there are bugs that may prevent these from updating everything properly. I would greatly appreciate any guidance on this (win10). I'm attaching the debug log files, and a word doc containing my command line output after trying to build XOD. Thank you.

[lerna-debug.log](https://github.com/xodio/xod/files/1570432/lerna-debug.log)

[command.docx](https://github.com/xodio/xod/files/1570440/command.docx)

nkrkv, 2017-12-19:
I see the problem is `babel-plugin-inline-import` can’t be found. The plugin is a dependency defined at `packages/xod-arduino/package.json`. It *should* be installed on running `yarn`.

Can you confirm that `yarn` (or `yarn install` which is the same) completes without errors? Also, check `./node_modules` directory. Do you see `babel-plugin-inline-import` there?

puttley, 2017-12-19:
Thank you. I do indeed have an error when running yarn install. I'm attaching my command window screen. Also, I do not have the babel-plugin-inline-import in my node_modules directory. If you could, please advise on how to fix. Thank you.

[yarn_fail.docx](https://github.com/xodio/xod/files/1572026/yarn_fail.docx)

puttley, 2017-12-19:
Here are last debug logs after trying to build.
[npm-debug.log](https://github.com/xodio/xod/files/1572046/npm-debug.log)
[lerna-debug.log](https://github.com/xodio/xod/files/1572047/lerna-debug.log)

nkrkv, 2017-12-19:
A build will fail until you have a successful `yarn install` run. So let’s focus on `yarn install`

1. Clone the xod repo from scratch to a new directory
2. Run `yarn install`
3. Literally, publish all console output here (use triple backticks to make it rendered as code, do not use docx)
4. Attach `*.log` you get

puttley, 2017-12-19:
Thank you. I cloned to a new directory, ran yarn install. It did not generate any log files - no *.log found in the new directory.
```
C:\Users\uttleyp\Documents\GitHub\xod_1\xod>yarn install
yarn install v1.3.2
[1/4] Resolving packages...
[2/4] Fetching packages...
error Command failed.
Exit code: 128
Command: git
Arguments: archive 07c72f8066b28ec2ea925865272528aaa9e047b5
Directory: C:\Users\uttleyp\scoop\apps\yarn\current\cache\v1\.tmp\8b529314dc73f63af2865eaa963cf20d
Output:

info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.

C:\Users\uttleyp\Documents\GitHub\xod_1\xod>
```

nkrkv, 2017-12-19:
Hmmm… yarn is unable to fetch a git dependency for some odd reason. Try:

```bash
yarn cache clean && yarn install
```

or

```bash
yarn upgrade && yarn install
```

puttley, 2017-12-19:
Report from yarn cache clean && yarn install:
```
C:\Users\uttleyp\Documents\GitHub\xod_1\xod>yarn cache clean && yarn install
yarn cache v1.3.2
success Cleared cache.
Done in 23.46s.
yarn install v1.3.2
[1/4] Resolving packages...
[2/4] Fetching packages...
error Command failed.
Exit code: 128
Command: git
Arguments: clone https://github.com/xodio/electron-builder C:\Users\uttleyp\scoop\apps\yarn\current\cache\v1\.tmp\8b529314dc73f63af2865eaa963cf20d
Directory: C:\Users\uttleyp\Documents\GitHub\xod_1\xod
Output:
Cloning into 'C:\Users\uttleyp\scoop\apps\yarn\current\cache\v1\.tmp\8b529314dc73f63af2865eaa963cf20d'...
git-lfs filter-process: git-lfs: command not found
fatal: The remote end hung up unexpectedly
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry the checkout with 'git checkout -f HEAD'
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.
```
Report from yarn upgrade && yarn install
```
C:\Users\uttleyp\Documents\GitHub\xod_1\xod>yarn upgrade && yarn install
yarn upgrade v1.3.2
[1/4] Resolving packages...
warning documentation > babel-preset-es2015@6.24.1: �  Thanks for using Babel: we recommend using babel-preset-env now: please read babeljs.io/env to update!
warning spectron > electron-chromedriver > electron-download > nugget > progress-stream > through2 > xtend > object-keys@0.4.0:
[2/4] Fetching packages...
error Command failed.
Exit code: 128
Command: git
Arguments: archive 07c72f8066b28ec2ea925865272528aaa9e047b5
Directory: C:\Users\uttleyp\scoop\apps\yarn\current\cache\v1\.tmp\8b529314dc73f63af2865eaa963cf20d
Output:

info Visit https://yarnpkg.com/en/docs/cli/upgrade for documentation about this command.
```

nkrkv, 2017-12-20:
Thanks. I think the core of the problem is in

```
git-lfs filter-process: git-lfs: command not found
fatal: The remote end hung up unexpectedly
warning: Clone succeeded, but checkout failed.
```

I’m not a Windows user and our devs are not Windows as well, so I’m unsure… But it looks like you have some kind of “incomplete” Git installation on your machine. It wants `git-lfs` but can’t find it. Can you find it on your PC?

```bash
$ which git-lfs
```

If not, here you can take it: https://git-lfs.github.com/

If doesn’t help could you try the same commands but with `--verbose` flag?

```bash
$ yarn --verbose cache clean && yarn --verbose install
$ yarn --verbose upgrade && yarn --verbose install
```

P.S> The whole situation is strange since the repo builds fine on Linux, Mac, our canary Windows laptop, and CI servers including Windows one.

puttley, 2017-12-20:
Thanks for all of your help, Victor. I got the git issue resolved by adding git to my Windows path. It was missing. Although now I am seeing other errors and warnings as shown below.
Question: Is there a chance this could this be caused by me being behind a company firewall, or by an antivirus SW?

```
$ yarn cache clean && yarn install
yarn cache v1.3.2
success Cleared cache.
Done in 24.38s.
yarn install v1.3.2
[1/4] Resolving packages...
[2/4] Fetching packages...
info fsevents@1.1.2: The platform "win32" is incompatible with this module.
info "fsevents@1.1.2" is an optional dependency and failed compatibility check. Excluding it from installation.
info 7zip-bin-linux@1.1.0: The platform "win32" is incompatible with this module.
info "7zip-bin-linux@1.1.0" is an optional dependency and failed compatibility check. Excluding it from installation.
info 7zip-bin-mac@1.0.1: The platform "win32" is incompatible with this module.
info "7zip-bin-mac@1.0.1" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client-browser > react-skylight@0.4.2" has inco
rrect peer dependency "react@^0.14.0 || ^15.0.1".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-autolink-text2@3.0.0" has unmet
peer dependency "react@^0.14.9 || ^15.3.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-autosuggest@9.3.2" has unmet pee
r dependency "react@>=0.14.7".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client-browser > react-event-listener@0.4.5" ha
s incorrect peer dependency "react@^0.14.0 || ^15.0.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client-browser > why-did-you-update@0.1.0" has
incorrect peer dependency "react@^15.0 || 0.14.x".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client-electron > chai-as-promised@6.0.0" has u
nmet peer dependency "chai@>= 2.1.2 < 4".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-codemirror@1.0.0" has unmet peer
 dependency "react@>=15.5 <16".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-codemirror@1.0.0" has unmet peer
 dependency "react-dom@>=15.5 <16".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-collapsible@2.0.3" has unmet pee
r dependency "prop-types@^15.5.8".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-collapsible@2.0.3" has unmet pee
r dependency "react@~15 || ~16".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-collapsible@2.0.3" has unmet pee
r dependency "react-dom@~15 || ~16".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-contextmenu@2.9.1" has unmet pee
r dependency "react@^0.14.0 || ^15.0.0 || ^16.0.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-contextmenu@2.9.1" has unmet pee
r dependency "react-dom@^0.14.0 || ^15.0.0 || ^16.0.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-contextmenu@2.9.1" has unmet pee
r dependency "prop-types@^15.0.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-custom-scroll@2.0.1" has unmet p
eer dependency "react@*".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-custom-scroll@2.0.1" has unmet p
eer dependency "react-dom@*".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-custom-scroll@2.0.1" has unmet p
eer dependency "prop-types@*".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-dnd@2.5.4" has unmet peer depend
ency "react@*".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-highlight-words@0.8.1" has unmet
 peer dependency "react@^0.14.0 || ^15.0.0 || ^16.0.0-0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-infinite@0.12.1" has unmet peer
dependency "react@^15.5.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-infinite@0.12.1" has unmet peer
dependency "react-dom@^15.5.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-infinite@0.12.1" has unmet peer
dependency "prop-types@^15.5.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-reflex@2.0.11" has unmet peer de
pendency "react@^16.0.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-reflex@2.0.11" has unmet peer de
pendency "react-dom@^16.0.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-sortable-hoc@0.6.7" has unmet pe
er dependency "react@^0.14.0 || ^15.0.0".
warning "workspace-aggregator-35416bf5-8588-43fe-a17d-29e9bf3b39a5 > xod-client > react-sortable-hoc@0.6.7" has unmet pe
er dependency "react-dom@^0.14.0 || ^15.0.0".
[... 214 lines cut ...]
    at loadEnv.then (C:\Users\uttleyp\Documents\GitHub\xod_1\xod\node_modules\electron-builder\src\cli\cli.ts:49:4)
    at runCallback (timers.js:666:20)
    at tryOnImmediate (timers.js:639:5)
    at processImmediate [as _immediateCallback] (timers.js:611:5)
[... cut ...]
    at Object.get [as argv] (C:\Users\uttleyp\Documents\GitHub\xod_1\xod\node_modules\electron-builder\node_modules\yarg
s\yargs.js:912:21)
    at Object.<anonymous> (C:\Users\uttleyp\Documents\GitHub\xod_1\xod\node_modules\electron-builder\src\cli\cli.ts:43:1
5)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)

uttleyp@A4333 MINGW64 ~/Documents/GitHub/xod_1/xod (master)
$
```

nkrkv, 2017-12-20:
OK, fine, now it looks like the problem is:

```
C:\Users\uttleyp\Documents\GitHub\xod_1\xod\node_modules\serialport\build\serialport.vcxproj(20,3): error MSB4019: The i
mported project "C:\Microsoft.Cpp.Default.props" was not found. Confirm that the path in the <Import> declaration is cor
rect, and that the file exists on disk.
```

The error itself is caused by Yarn/Node unable to natively build the `serialport` module. From what I see, it looks like on Windows it is build by MSVC toolchain which creates temporary files on `C:\`. Do you have a write access to the disk C root?

puttley, 2017-12-20:
Hi Victor, thanks for sticking with me on this. I checked my user permissions - I had those changed, and am now set at full control so I do have write access to the disk C root. Can you be more specific about how to confirm the path in the import declaration?
I re-cloned and tried to install yarn again. Here is the result. Very frustrating...thanks.
```
uttleyp@A4333 MINGW64 ~/Documents/GitHub/xod_1/xod (master)
$ yarn
yarn install v1.3.2
[1/4] Resolving packages...
[2/4] Fetching packages...
info fsevents@1.1.2: The platform "win32" is incompatible with this module.
info "fsevents@1.1.2" is an optional dependency and failed compatibility check. Excluding it from installation.
info 7zip-bin-linux@1.1.0: The platform "win32" is incompatible with this module.
info "7zip-bin-linux@1.1.0" is an optional dependency and failed compatibility check. Excluding it from installation.
info 7zip-bin-mac@1.0.1: The platform "win32" is incompatible with this module.
info "7zip-bin-mac@1.0.1" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client-browser > react-skylight@0.4.2" has inco
rrect peer dependency "react@^0.14.0 || ^15.0.1".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-autolink-text2@3.0.0" has unmet
peer dependency "react@^0.14.9 || ^15.3.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-autosuggest@9.3.2" has unmet pee
r dependency "react@>=0.14.7".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client-browser > react-event-listener@0.4.5" ha
s incorrect peer dependency "react@^0.14.0 || ^15.0.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client-browser > why-did-you-update@0.1.0" has
incorrect peer dependency "react@^15.0 || 0.14.x".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client-electron > chai-as-promised@6.0.0" has u
nmet peer dependency "chai@>= 2.1.2 < 4".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-codemirror@1.0.0" has unmet peer
 dependency "react@>=15.5 <16".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-codemirror@1.0.0" has unmet peer
 dependency "react-dom@>=15.5 <16".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-collapsible@2.0.3" has unmet pee
r dependency "prop-types@^15.5.8".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-collapsible@2.0.3" has unmet pee
r dependency "react@~15 || ~16".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-collapsible@2.0.3" has unmet pee
r dependency "react-dom@~15 || ~16".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-contextmenu@2.9.1" has unmet pee
r dependency "react@^0.14.0 || ^15.0.0 || ^16.0.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-contextmenu@2.9.1" has unmet pee
r dependency "react-dom@^0.14.0 || ^15.0.0 || ^16.0.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-contextmenu@2.9.1" has unmet pee
r dependency "prop-types@^15.0.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-custom-scroll@2.0.1" has unmet p
eer dependency "react@*".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-custom-scroll@2.0.1" has unmet p
eer dependency "react-dom@*".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-custom-scroll@2.0.1" has unmet p
eer dependency "prop-types@*".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-dnd@2.5.4" has unmet peer depend
ency "react@*".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-highlight-words@0.8.1" has unmet
 peer dependency "react@^0.14.0 || ^15.0.0 || ^16.0.0-0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-infinite@0.12.1" has unmet peer
dependency "react@^15.5.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-infinite@0.12.1" has unmet peer
dependency "react-dom@^15.5.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-infinite@0.12.1" has unmet peer
dependency "prop-types@^15.5.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-reflex@2.0.11" has unmet peer de
pendency "react@^16.0.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-reflex@2.0.11" has unmet peer de
pendency "react-dom@^16.0.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-sortable-hoc@0.6.7" has unmet pe
er dependency "react@^0.14.0 || ^15.0.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > react-sortable-hoc@0.6.7" has unmet pe
er dependency "react-dom@^0.14.0 || ^15.0.0".
warning "workspace-aggregator-0e65b9b4-8f5e-4542-87e5-a621a217ae08 > xod-client > recompose@0.25.1" has unmet peer depen
dency "react@^0.14.0 || ^15.0.0 || ^16.0.0".
[... cut ...]
    at Object.get [as argv] (C:\Users\uttleyp\Documents\GitHub\xod_1\xod\node_modules\electron-builder\node_modules\yarg
s\yargs.js:912:21)
    at Object.<anonymous> (C:\Users\uttleyp\Documents\GitHub\xod_1\xod\node_modules\electron-builder\src\cli\cli.ts:43:1
5)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)

uttleyp@A4333 MINGW64 ~/Documents/GitHub/xod_1/xod (master)
$
```

nkrkv, 2017-12-21:
Oh! I think the truth is out there.

```
error MSB8020: The build to
ols for Visual Studio 2010 (Platform Toolset = 'v100') cannot be found. To build using the v100 build tools, please inst
all Visual Studio 2010 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the
Project menu or right-click the solution, and then selecting "Retarget solution". [C:\Users\uttleyp\Documents\GitHub\xod
_1\xod\node_modules\serialport\build\serialport.vcxproj]
```

Serialport module is native one, written in C++. Do you have the [Visual Studio build tools](http://landinghub.visualstudio.com/visual-cpp-build-tools) installed? They are used on Windows to build native modules.

puttley, 2017-12-21:
Victor -- Success !!

I had VS 2015 C++ tools installed. I updated to the 2017 VS build tools which fixed everything. Thank you so much for all of your help...much appreciated.

nkrkv, 2017-12-21:
You’re welcome ;)

closed 2017-12-21 by nkrkv