xodio/xod #1110: Grove DHT22 (AM2302) and I2C LCD do not work together

nkrkv, 2018-03-12:
The problem [reported on the forum](https://forum.xod.io/t/problem-with-dht22-am2302-grove/536/2)

### Steps to reproduce

1. Build a circuit with [Grove DHT22](https://www.seeedstudio.com/Grove-Temperature&Humidity-Sensor-Pro-p-838.html) on port D2 and a text I2C display
2. Make a program to show temperature and humidity from the sensor on the screen
3. Upload

### Expected behavior

LCD shows actual values

### Actual behavior

LCD shows zeros for both T and RH

### Details

It was verified another very similar AM2301 sensor works fine with an I²C display, but it looks like AM2302 differs somehow.

Help from one who has following is required:

- Grove DHT22
- I²C Text LCD
- C++ knowledge and Arduino debug skills

The following yet to be done:

- Confirm or decline the bug
- Find the root cause

maroelawerner, 2018-10-17:
I have the same problem. Tested DHT22 sensor using the Arduino IDE and the DHTlib and it works well. Using XOD it just shows values of 0.00.

evgenykochetkov, 2018-10-17:
@maroelawerner, did you use XOD v0.25.0?

maroelawerner, 2018-10-18:
I have upgraded to V0.25.0 and tried out you recommendation. 
Still the same.
BTW. The original project had a square wave generator connected to the UPD.

evgenykochetkov, 2018-10-18:
Then please also update to v0.25.0. Among [other nice stuff](https://github.com/xodio/xod/releases/tag/v0.25.0), it comes with fixes for DHT11 and DHT2x nodes.

maroelawerner, 2018-10-20:
Hi, 

I'm using an Arduino clone and an AM2302. 

I have attached some pictures showing you the details of the setup. 
I have also have pictures of the oscilloscope traces. With the XOD download, there is just a high level after a short low. With the Arduino IDE download, that works without changing the hardware at all, you can see the exchange of data taking place. It looks like the request to send data is not handled properly by the XOD code.

[temp-humid.xodball.zip](https://github.com/xodio/xod/files/2498084/temp-humid.xodball.zip)
[image]
[image]
[image]
[image]
[image]

maroelawerner, 2018-10-20:
missed one picture
[image]

evgenykochetkov, 2018-10-23:
@maroelawerner, thanks for the detailed report!

That still looks a lot like `xod/common-hardware/dht2x-thermometer` from v0.24.x🤔
Don't  you by any chance have an old version of `xod/common-hardware` in your `~/xod/__lib__`?

nkrkv, 2019-10-04:
Should be fixed by #1853

closed 2019-10-04 by nkrkv