xodio/xod #1114: Legalize parametric patch names

nkrkv, 2018-03-12:
### Rationale

To support generics we should allow patch names which define the type(s) they specialize for:

- `if-else (number)`
- `if-else (string)`
- `cast (boolean, pulse)`

To be more friendly than.A$Perl<> (follow basic visual design principles) I also suggest enforcing a space before the left parenthesis and after a comma.

### Acceptance criteria

The following are legal and can be created via IDE:

- [ ] `if-else (number)`
- [ ] `cast (boolean, pulse)`

The following is not:

- [ ] `if-else (number`
- [ ] `if-else string)`
- [ ] `if-else ()`
- [ ] `if-else (number) string`

When creating a patch IDE automatically normalizes spaces:

- [ ] `if-else(number)` → `if-else (number)`
- [ ] `if-else   (  number  )` → `if-else (number)`
- [ ] `cast (boolean ,pulse)` → `cast (boolean, pulse)`

### Out of scope

Any special meaning for the patches with parenthesis in their name.

brusherru, 2018-03-12:
Spaces will be ugly in the URLs for library docs.
So maybe keep patch paths without spaces, but for better readability, we could render it with spaces in the GUI?

nkrkv, 2018-03-13:
URLs :thinking: Very good point!

OK, I agree. Will try to use GUI-only spaces or better half-spaces or quarter-spaces. Whatever will work better visually. I’ll edit the issue body.

PR #1128 (2018-03-15): Legalize parametric patch names
It closes #1114.

Also tweaked rename popup. Now input already contains current patch name and autoselected.
closed 2018-03-15 by brusherru