xodio/xod #1116: As a xoder I want to create abstract nodes so that I can define interface for future specializations

nkrkv, 2018-03-12:
### Rationale

An end-user wants to simply place `if-else` node regardless of the concrete type of its arguments, not the `if-else (number)` or `if-else (string)`. To make it possible we introduce a concept of the abstract node.

Later it will resolve to a particular specialization or throw an error. But before that, we should be able to create it.

To make a node abstract, a xoder places `abstract` marker node on the patch node and at least one terminal of generic type `t1`, `t2`, `t3`. If generic type terminals are absent, that’s an error which is shown in a tooltip like it is done for `variadic-*` nodes.

### Acceptance criteria

- [ ] I can place the `abstract` marker node
- [ ] I can place `input (t1)`, `output (t1)`, `input (t2)`, `output (t2)`, `input (t3)`, `output (t3)` terminal nodes
- [ ] The generic terminal nodes render the same way as regular terminals do and they are colored gray
- [ ] The generic pins render the same way as regular pins do and they are colored gray
- [ ] If a patch node has `abstract` and no inputs of `t1`, the marker is drawn in red and shows “At least one generic `input (t1)` terminal is required”
- [ ] If a patch node has `abstract`, `input (t1)`, `input (t3)`, but no `input (t2)`, the marker errors with “The generic inputs t1, t2, t3 should be employed sequentially”
- [ ] Documentation boxes show “generic (t1)” as the type label for generic pins

### Out of scope

Generic pins linking may not be supported

evgenykochetkov, 2018-03-13:
About validation: do we allow generic terminals when there is no `abstract` marker?

nkrkv, 2018-03-13:
Sure. It can be a patch node which composes other generic nodes.

PR #1132 (2018-03-16): Allow to create abstract nodes
Closes #1116
closed 2018-03-16 by evgenykochetkov