xodio/xod #1221: Can’t unbind values from generics

nkrkv, 2018-05-18:
### Environment

- Scope: desktop IDE, browser IDE, CLI
- Version: master, 079ab358e6d25eef913bd4f00ae2df7613c36ec6

### Steps to reproduce

1. Place an `if-else` node
2. Bind `42` to `T`
3. Clear the value for `T` and press Enter

### Expected behavior

The node switches to the initial state, that is nothing bound

### Actual behavior

`T` pin goes red with an error “Value undefined bound to T is invalid”

brusherru, 2018-05-18:
Also, we have a problem with binding some bytes and "almost valid" types.
Try to bind `32d` or `true` in the generic pin.

nkrkv, 2018-05-18:
Are you sure they’re related? If not, that should be a separate issue.

commit cfc74027 (2018-05-18): fix(xod-client): allow unbinding generic values in inspector

Fixes #1221
commit 4a4d2182 (2018-05-21): fix(xod-client): allow unbinding generic values in inspector

Fixes #1221
PR #1226 (2018-05-21): Allow unbinding generic values in inspector
Fixes #1221

Problem was introduced in #1192 and does not affect `0.20.x`.
closed 2018-05-21 by evgenykochetkov