xodio/xod #1239: `Unsafe` suffix for functions has different meanings

evgenykochetkov, 2018-05-23:
We have several `*Unsafe` functions defined in our project

```
$ ag 'Unsafe ::'
packages/xod-project/src/patch.js
248:  'getNodeByIdUnsafe :: NodeId -> Patch -> Node',
499:  'getLinkByIdUnsafe :: LinkId -> Patch -> Link',
750:  'getCommentByIdUnsafe :: CommentId -> Patch -> Comment',

packages/xod-project/src/xodball.js
35:  'fromXodballDataUnsafe :: Object -> Project',

packages/xod-project/src/project.js
318:  'getPatchByPathUnsafe :: PatchPath -> Project -> Patch',
700:  'assocPatchUnsafe :: PatchPath -> Patch -> Project -> Project',
734:  'assocPatchListUnsafe :: [Patch] -> Project -> Project',
948:  'getPatchByNodeIdUnsafe :: NodeId -> Patch -> Project -> Patch',
1105:  'changeNodeTypeUnsafe :: PatchPath -> NodeId -> PatchPath -> Project -> Project',

packages/xod-project/src/utils.js
214:  'getTypeFromLiteralUnsafe :: DataValue -> DataType',

packages/xod-arduino/src/transpiler.js
189:  'getNodePinsUnsafe :: Node -> Project -> [Pin]', — throws
```

For most of them it means "try doing stuff and throw an exception if it's invalid".
But for others, like `assocPatchUnsafe`, it means "just do that thing and don't care if the result is not 100% valid"(resulting project could have some dead references or links with incompatible types, but our code won't crash)

- Leave `Unsafe` suffix only for functions that will throw exceptions
- For others, invent another suffix(?) and rename them
   - `assocPatchUnsafe :: PatchPath -> Patch -> Project -> Project`
   - `assocPatchListUnsafe :: [Patch] -> Project -> Project`
   - `changeNodeTypeUnsafe :: PatchPath -> NodeId -> PatchPath -> Project -> Project`

nkrkv, 2018-05-23:
Possible variants:

1)

- `xxxExn` is for exception throwing (like in ReasonML)
- `xxxUnsafe` is for those who can possibly violate the integrity

2)

- `xxxUnsafe` for exceptions
- `xxxForced` for integrity

3)

Never care about integrity on fly. Check it only in `validateXXX`.

brusherru, 2018-08-13:
In PR #1402 I did that we don't care about integrity on the fly (check it only in `validateXXX`).
So `xxxUnsafe` only for exceptions, that could be caused by exploding Maybies or Eithers.

PR #1402 (2018-08-13): Get rid of integrity validations
It closes #1369 and closes #1239
closed 2018-08-13 by brusherru