xodio/xod #1335: Eliminate `any` in generated C++

nkrkv, 2018-07-06:
### Rationale

`xod/core/any` is a widespread node in generated C++ especially if it participates in `ERR` convergence where variadic `any`s are used extensively. Each instance consumes 3 bytes of global RAM although the semantics is trivial and can be baked right into the generated code

In an [example program](https://github.com/xodio/xod/files/2170921/mbot_lib23123123.zip) (patch 123) compilation gives:

```
AVR Memory Usage
----------------
Device: atmega328p

Program:   35184 bytes (107.4% Full)
(.text + .data + .bootloader)

Data:       1833 bytes (89.5% Full)
(.data + .bss + .noinit)
```

There are 93 `any`s there consuming at least 279 bytes of RAM (`.bss`) and a considerable amount of flash (`.text`) for code in `runTransaction`.

### How to optimize

Instead of placing `any` nodes on the regular basis we can:

1. Skip pasting of their implementation completely
2. Skip generation of evaluation blocks in `evaluate`¹
3. When an upstream node wants to mark a downstream `any` as dirty, follow the cascade down and mark the first non-`any` node as dirty directly

¹
```cpp
    { // xod__core__any #268
        if (node_268.isNodeDirty) {
            XOD_TRACE_F("Eval node #");
            XOD_TRACE_LN(268);

            xod__core__any::ContextObject ctxObj;
            ctxObj._node = &node_268;

            // copy data from upstream nodes into context
            ctxObj._input_IN1 = node_261.output_OUT;
            ctxObj._input_IN2 = node_266.output_OUT;

            ctxObj._isInputDirty_IN1 = node_261.isOutputDirty_OUT;
            ctxObj._isInputDirty_IN2 = node_266.isOutputDirty_OUT;

            xod__core__any::evaluate(&ctxObj);

            // mark downstream nodes dirty
            node_282.isNodeDirty |= node_268.isOutputDirty_OUT;
        }
    }
```

evgenykochetkov, 2018-07-11:
How about just deprecating `any` and allowing linking multiple pulses to a single input?

nkrkv, 2018-07-11:
I thought about it, but that seems like a separate issue or not an issue at all. It does not matter *how* we got a situation in which multiple pulses are coming into a single input. What matters is that we can squash the node for optimization.

When we allow multiple incoming pulses one of graph processing stages can replace it with an equivalent subgraph based on `any`. So that the transpiler can still rely on the fact that any input can have at most one incoming link.