xodio/xod #1354: Make watches and constants resizable

nkrkv, 2018-07-12:
### Rationale

It’s often desired that watches and constants may have a quite long label which can’t fit one slot. Sometimes it makes them just unusable. Let xoders resize the watches and constants with the same UX as they have when resizing comments.

### Acceptance criteria

- [ ] I can drag the bottom right corner of any `watch` or `xod/core/constant-*` to adjust the node’s width and height (in steps of a slot)
- [ ] The label aligns to center/middle regardless of size

### How to implement

Make the ability to resize a `Node` attribute. Having that we can later easily transform the comments as a separate issue to nodes with a special borderless look.

evgenykochetkov, 2018-07-12:
Resizing a watch node while in debug mode will stop debug mode(because the patch has changed).
I think we'll need to address that as well.

nkrkv, 2018-07-17:
Addressed in #1368

PR #1375 (2018-07-18): Make constant & watch nodes resizable
It closes #1354.
closed 2018-07-18 by brusherru