xodio/xod #1385: Multiline strings in inspector and in constant Nodes look

brusherru, 2018-07-19:
### Rationale
Sometimes User needed to write a multiline string (EG, for an HTTP request).
Now users should write "\r\n" symbols. And it looks like a long string, instead of real new lines.

Let's add first extended Widget for a String type.

The user can click on the icon next to input and it will show a textarea instead. New lines will be automatically replaced with `\r\n` (it could be a switchable option: 'no new lines', '\r\n', '\n'). After losing focus or clicking "OK" button value will be placed inside input with `\r\n`s.

Also, resizable Nodes (like `constant-string`) should show up multiline labels/values.

### Prerequisites
Probably, we have to refactor the widget system 🙈 

### Acceptance criteria
- [ ] I can click an icon next to String input and textarea will be shown
- [ ] I can enter multiline text and close textarea by clicking somewhere outside
- [ ] I'll see the same text in the input with `\r\n` symbols
- [ ] If I do this with `constant-string` Node and resize it — I'll see multiline text on the Node

### Out of scope
Multiline values in Watch nodes

nkrkv, 2018-07-19:
:+1: 

Not sure about `\r\n` though. Sometimes just `\n` is desirable. I think a user should have a feature to switch between CR and CR+LF somewhere near the textarea.

evgenykochetkov, 2018-07-20:
> Probably, we have to refactor the widget system

Awesome 👍🏻

brusherru, 2018-07-20:
>> I think a user should have a feature to switch...

Yep :)

>> it could be a switchable option: 'no new lines', '\r\n', '\n'