xodio/xod #1388: Pan patch on scroll events

nkrkv, 2018-07-25:
### Rationale

When using a laptop with a dumb touchpad, panning the patchboard can be tricky as it requires some acrobatics from your fingers. In contrast to desktop+mouse laptops support touchpad gestures existing to solve such problems. Notably, scrolling usually might be done with two-finger swiping.

Also, as noted by users, many of them expect that the mouse wheel will scroll the patch board as well.

XOD ignores the scrolling events and that makes it feel non-native. We can handle the vertical and horizontal scrolling events generated by a touchpad or mouse to make UI more intuitive.

### Acceptance criteria

* [ ] I can pan using the two-finger swipe gesture on a touchpad configured to use this gesture for scrolling
* [ ] I can scroll the mouse wheel to pan the patch board vertically
* [ ] I can hold the Alt key and scroll the mouse wheel and the patch board will scroll horizontally

### TODO

Is the `Alt` key the most popular to alter the scrolling direction? Check graphical editors and CADs.

joepie91, 2018-08-21:
On certain systems, this is actually a *breaking issue*; it's common for trackpoint users to configure their system such that holding the middle mouse button + moving translates to scroll events, as there is no other way to express scrolling with a trackpoint.

Given that the only way to currently pan the workspace is by using that same combination (middle mouse button + moving mouse), it is *literally impossible* to pan the view on these systems, which includes many professional laptops like Thinkpads :(

Mapping scroll events (both horizontally and vertically!) would fix not only the trackpad issue, but also make the application usable on Trackpoint-based systems.

nkrkv, 2018-08-22:
@joepie91 That’s very valuable information. I’ve written this issue after trying XOD on my Thinkpad laptop. I’ve failed to find any way to pan as you mention, but I had no idea *why* it happens. Now I know, thank you.

The fix is on its way in #1412. Requires a comprehensive testing though.

joepie91, 2018-08-22:
@nkrkv Would it be helpful if I did some 'end-user testing' of the patched version (in Electron) on my laptop some time this week?

nkrkv, 2018-08-22:
Yes, it would be very valuable! Please test against the `feat-1388-pan-patch-on-scroll` branch:

```
git checkout feat-1388-pan-patch-on-scroll && yarn && yarn build
```

We’ll be glad to hear any feedback. Thanks.

PR #1412 (2018-08-29): Scroll patch with mouse wheel or trackpad
It will close #1388
closed 2018-08-29 by brusherru