xodio/xod #1423: Inspector scrolling is broken

joepie91, 2018-08-22:
### Environment

- Scope: Desktop IDE
- Version: "0.23.0" from repository master
- OS: NixOS 18.03
- Machine: i5, 8GB RAM

### Steps to reproduce

Start a project; add a node with many options such that there are too many to show in the inspector pane (or resize the inspector pane to *cause* this problem), and select the node.

### Expected behavior

Options overflow and a scrollbar becomes available.

### Actual behavior

The options will run off the screen without any scrolling option available; the pane is actually explicitly set to `overflow: hidden`. It's not possible to normally scroll the inspector pane, and the pane needs to be resized to make all options visible.

### Console output

N/A.

### How to fix 

Set `overflow: hidden` to `overflow: auto` instead; however, the rule currently exists on `.reflex-container .reflex-element`, which seems to be a more general rule than just for the inspector. The project browser seems to solve this with a `.custom-scroll` wrapper?

The changelog suggests that this was fixed once in 0.15.0. I guess this is a regression?

nkrkv, 2018-08-22:
Hello and thank you for the report. Can’t reproduce it as is on my machine (Arch Linux, Gnome):

[image]

The inspector (as the project browser and many other panes) indeed uses a custom scroll component which pushes the native scrollbar out of the visible area and draws a styled one instead.

I can easily scroll it with the wheel as shown on the screencast. And I can drag the scroll gripper as well. The latter requires pixel-perfect pointing and, yes, seems a bit buggy. But overall the scrolling works.

Can you confirm that you don’t have the scrollbar at all and have no chance to scroll at all (even with the wheel)? If so, would you tell what your DM is (Gnome, KDE, XFCE, i3, whatever?)

joepie91, 2018-08-22:
Bizarrely, I can't reproduce it either today after having restarted the application (due to having messed with developer tools a bit and wanting to reset to a clean state). I could reliably reproduce it yesterday, though, and I'm fairly sure that the `custom-scroll` wasn't there yesterday in the Inspector pane, only in the project browser.

Is there some kind of logic that determines whether to use a `custom-scroll` wrapper or not, that might have different behaviour depending on how the application is initialized? Eg. only being added if the window is below certain dimensions?

If not, I'll have to wait until the problem reoccurs and further debug at that point... I did run into another scrolling bug where it jumps to the top/bottom of the content if you scroll too fast near the edges, but I should probably open a separate issue for that once I've figured out the details, and I don't think it's directly related to this one.

(I'm running KDE Plasma, by the way.)

nkrkv, 2018-08-22:
That’s weird

> Is there some kind of logic that determines whether to use a custom-scroll wrapper or not

No AFAIK, it is always there.

> but I should probably open a separate issue for that once I've figured out the details

Feel free to re-open this issue if it makes sense. I’m closing it for now.

closed 2018-08-22 by nkrkv