xodio/xod #1432: Outputs are reversed in helpboxes sometimes

nkrkv, 2018-09-05:
- Version: 0.23.0
- [Reported](https://forum.xod.io/t/bug-in-built-in-help/1098) on the forum, I confirm it

### Steps to reproduce

Open the helpbox for `xod/core/nth-input(string)`

### Expected behavior

The outputs are listed (from bottom to top) in order:

- `OUT`
- `_`

### Actual behavior

They are listed in reverse

### Details

The same behavior reproduces for some nodes (e.g. `xod/stream/pass-from-sequence`), but not for others (e.g. `xod/bits/number-to-u32`). Perhaps the bug affects nodes with auto-generated pin labels only.

brusherru, 2018-10-18:
[image]
Looks like this is not actual anymore... or I did not understand the bug 🤔

nkrkv, 2018-10-18:
> The same behavior reproduces for some nodes (e.g. xod/stream/pass-from-sequence), **but not for others** (e.g. xod/bits/number-to-u32).

evgenykochetkov, 2018-10-23:
@brusherru, it is solved by something like this:

```diff
From 91090405f17cee7e9f3646d698a2b244bb15b87b Mon Sep 17 00:00:00 2001
From: Evgeny Kochetkov <evgeny.kochetkov@me.com>
Date: Wed, 12 Sep 2018 12:40:44 +0300
Subject: [PATCH] fix(xod-client): ensure pin order in patch docs

Fixes #1432
---
 packages/xod-client/src/editor/components/PatchDocs.jsx | 1 +
 1 file changed, 1 insertion(+)

diff --git a/packages/xod-client/src/editor/components/PatchDocs.jsx b/packages/xod-client/src/editor/components/PatchDocs.jsx
index 0d70635a..c8a99fe3 100644
--- a/packages/xod-client/src/editor/components/PatchDocs.jsx
+++ b/packages/xod-client/src/editor/components/PatchDocs.jsx
@@ -123,6 +123,7 @@ const PatchDocs = ({ patch, minimal }) => {
   )(patch);
 
   const [inputPins, outputPins] = R.compose(
+    R.map(R.sortBy(XP.getPinOrder)),
     R.partition(XP.isInputPin),
     XP.normalizeEmptyPinLabels,
     R.map(pin =>
-- 
2.15.0

```

commit 4feaf1fc (2018-10-30): fix(xod-client): ensure pin order in patch docs

Fixes #1432
commit 720a6043 (2018-11-06): fix(xod-client): ensure pin order in patch docs

Fixes #1432
closed 2018-11-01 by brusherru