xodio/xod #1439: Support built-in pull-up resistors

nkrkv, 2018-09-10:
### Rationale

All Arduino-compatible boards I know have built-in pull-up resistors on their pins. They make it simple to connect mechanical switches which are left with a floating signal in one of their states.

With the current standard library, a xoder is forced to use a dedicated hardware pull-up/pull-down resistor which is not always convenient. Let’s extend the standard library a bit to make it more pleasant to work with.

### Node adjustments

The `xod/gpio/digital-read` patch node should get one extra input:

- [in] PULL :: Boolean — Pull-up. Enables the internal built-in pull-up resistor while reading which forces the resulting signal to be `True` if the wire on the port is floating (disconnected).

The `xod/common-hardware/button` patch node should bind the `PULL` value to `True`.

### Acceptance criteria

- [ ] `digital-read` is updated
- [ ] `button` is updated, its description is adjusted accordingly

awgrover, 2018-12-30:
I question whether this node, `digital-read` is where the pull-up setting should go.

`pull-up` is usually an attribute/setting of the pin, rather than something that changes very often. So, I think that `set-pull-up` should be a separate node.

I started looking at adding `pull` to `digital-read`, and it bothered me that we'd have to check the `pull-up` input everytime:

```
oid evaluate(Context ctx) {
    if (!isInputDirty<input_UPD>(ctx))
        return;

    const uint8_t port = getValue<input_PORT>(ctx);
    if (!isValidDigitalPort(port)) {
        emitValue<output_ERR>(ctx, 1);
        return;
    }

    if (!isInputDirty<input_PULL>(ctx)) {
        const bool pull_up = getValue<input_PULL>(ctx);
        if (pull_up) {
            ::pinMode(port, INPUT_PULLUP);
        else {
            ::pinMode(port, INPUT);
        }
    }
    emitValue<output_SIG>(ctx, ::digitalRead(port));
    emitValue<output_DONE>(ctx, 1);
}
```

I have implemented the change, as https://github.com/awgrover/xod/commit/fa3094ff312bf103685267b6272fb094f9b59d55 . I haven't made the pull yet, as I haven't actually tested the code.

nkrkv, 2019-01-09:
I don’t think the check is a big deal. It should produce a very little performance penalty, but on the other side, it makes things more accessible to [Pete](https://github.com/xodio/xod/wiki/Pete).

PR #1846 (2019-09-10): Add digital-read with pull-up
It closes #1439.
closed 2019-09-10 by brusherru