xodio/xod #1530: Undo/Redo works not stable on Windows

brusherru, 2018-11-07:
### Environment
- Scope: desktop IDE
- Version: 0.25.2
- OS: Windows 10

Undo/redo works fine for moving nodes or adding new node from Project Browser, but when User adds a new node using `i` hotkey it makes undo/redo works strange (need to click hotkey few times).

### Steps to reproduce
1. Open XOD IDE
2. Click `i` hotkey
3. Type "clock" and press "enter"
4. Move the node
5. Press Ctrl+Z

### Expected behavior
Node is moved back.

### Actual behavior
Nothing happened.
If click few more times it will do undo.
The same thing with "redo".

Also, if you click "undo" in menu item it will react instantly.

I did a small investigation, place `console.log` in accelerator function and in the Editor's command function. The last one was not called at all, accelerator function called only after few clicks on the hotkey.

Cesar-S, 2018-11-22:
hi, in Windows solved. Thanks

nkrkv, 2018-11-27:
Closed by #1556

PR #1556 (2018-11-21): Improve undo/redo hotkeys reliability
Closes #1530
closed 2018-11-27 by nkrkv