xodio/xod #1532: Verify  Guide - Custom Types

Cesar-S, 2018-11-09:
### Expected behavior

Should it be possible to connect connect `output-self` node to `output-time`?

### Actual behavior

Following the tutorial at the end you can not connect "`time`" with "`unpack`".

Realized the patch "`time`" (https://xod.io/docs/guide/custom-types/time.patch.png) 

It is not possible to connect it to "`unpack`" (https://xod.io/docs/guide/custom-types/unpack-example.patch.png) ,instead of `output-self` you must use `output-time`

nkrkv, 2018-11-12:
To me, everything seems right. The `output-self` node is a marker, not a terminal actually. It has no inputs and no outputs. The sole purpose of `output-self` on a patch is to tell XOD that this patch defines a new type. XOD will generate the `output-time` and `input-time` as a consequence of `output-self` placement.

The example used in the guide is not fake and it is an actual project which you can download here https://xod.io/docs/guide/custom-types/time.xodball. See the `unpack-example` patch. There `time` is linked with `unpack`.

If I do not understand you correctly, and there’s indeed a bug out there, feel free to reopen the issue.

Cesar-S, 2018-11-12:
Dear @nkrkv  , eh downloaded the example and it works correctly, but I have the error in unpack.

I open another XOD, I create two patches, in a copied node "time" and when I want to copy unpack there is an error.

"Invalid paste
The clipboard contains references to missing patches: @ / input-time.
Try to install libraries which provide the required patches. "

For some reason I do not have Input-time and output-time in my libraries.

The error is that by not having them use input-datatime

Cesar-S, 2018-11-12:
`As you might notice, at the moment you place an output-self node on a constructor patch two new terminal nodes are automatically generated next to the constructor patch. In our example, they are input-time and output-time.`

I found my error, translation or reading problem
thank you for your time excuse me

nkrkv, 2018-11-12:
Oh, the message is indeed misleading. I’m going to rephrase the issue and promote it to a tweak.

Cesar-S, 2018-11-12:
Something to add that might be useful, when I did the tutorial look for the xodball I did not find this link [time.xodball] (https://xod.io/docs/guide/custom-types/time.xodball)
I've looked again and it's not there.

With a little searching I come here [github (custom-types)] (https://github.com/xodio/xod-docs/tree/master/docs/guide/custom-types) if the files are there.

PR #1541 (2018-11-13): Clarify message about bad references in paste
Closes #1532
closed 2018-11-13 by nkrkv