xodio/xod #1552: Select all, works just by pressing "A"

Cesar-S, 2018-11-17:
### Expected behavior

To select all the content of the patch, Ctrl + A must be used.

### Actual behavior

- Using  the A key, select all the content.

- Using  Ctrl + A, select all the content.

Conclusion, Ctrl is not necessary, to use the shortcut.

nkrkv, 2018-11-19:
Although it is an interesting suggestion it is a controversial feature which should be discussed on the [forum](https://forum.xod.io) first. I cannot accept it to the backlog as is at its current state, so I’m closing the issue for now. It might be re-opened in the future if the community accepts the idea.

closed 2018-11-19 by nkrkv

nkrkv, 2018-11-19:
:astonished: My bad, sorry. I thought the proposal was to change from Ctrl+A to A (which is interesting albeit controversial).

Definitely a bug.

Cesar-S, 2018-11-23:
Solved. Thanks

PR #1554 (2018-11-21): Fix "Select All" hotkey in Desktop IDE
It fixes #1552