xodio/xod #1584: Sine wave degrades in performance over time

nkrkv, 2018-12-06:
Reported on the [forum](https://forum.xod.io/t/output-problem-with-while-c/1402/5).

### Environment

* Version: 0.25.3
* Board: Arduino Nano

### Steps to reproduce

1.  Upload a simple patch with `sine-wave` linked to `led` with 1 Hz frequency
2.  Wait a night

### Expected behavior

The LED keeps pulsing at 1 Hz

### Actual behavior

The frequency becomes much lower, by an order of magnitude. The longer you wait, the slower LED pulses. It slowly degrades 2×, 3×, 4×, 5×, etc

### Notes

- Perhaps, other waves are affected as well
- Did not check for it, but it can be due to declining floating point arithmetics when numbers become large

evgenykochetkov, 2018-12-06:
Should we try something like https://github.com/FastLED/FastLED/blob/master/lib8tion/trig8.h ?

nkrkv, 2018-12-06:
Maybe, but first, we should find and understand the root cause.