xodio/xod #1609: Transpiling error - implementation for @/node no found

Cesar-S, 2018-12-28:
## Bug Report

Node implementation error 

[image]

### Environment

* Scope: desktop IDE
* Version: 0.26.0
* OS: Windows 10

### Steps to reproduce

When compiling a main node the error occurs, when compiling the content of the node everything works fine.

[405xB.zip](https://github.com/xodio/xod/files/2714011/405xB.zip)

evgenykochetkov, 2018-12-28:
Hi. Thanks for the report!

For some reason, there is a `patch.cpp` file in `main`'s patch folder. Deleting it solved the issue.

I wonder how it got there. Could you please describe how you created that patch?

Cesar-S, 2018-12-28:
> Could you please describe how you created that patch?

Hi, from scratch, all 4051 / a / b, I also made a new patch and agrue node 4051.
I repeat to see if it happens again, but also in the same project I have 4052 / a / b and the same error
A 4053 that is simple there is no error.

4051 is only part of the project

Cesar-S, 2018-12-28:
Conclusion: to no-implemented-in-xod, it will never be eliminated, only if it is done manually

evgenykochetkov, 2018-12-28:
Made a quick fix that will be released soon: https://github.com/xodio/xod/pull/1616
Filed issue about "stranded" attachments here: https://github.com/xodio/xod/issues/1618, it will be fixed a bit later

closed 2019-01-21 by Cesar-S