xodio/xod #1631: Adjust tutorial project to introduce simulation, watches, and tweaks

nkrkv, 2019-01-15:
### Rationale

The interactive features of XOD make it much friendlier to newcomers compared to the traditional Arduino C++. They deserve a few dedicated lessons in the tutorial project.

### TOC changes proposed

TODO

### Acceptance criteria

* [ ] The tutorial project is changed
* [ ] The changes are reflected at https://github.com/xodio/xod-docs

donovaly, 2019-01-21:
Please also add a new patch in the tutorial that describes the tabtest feature (introduced in XOD 0.26). This feature is in my opinion very important and useful.

nkrkv, 2019-01-21:
:100: agree. At its current stage, the tutorial ends before it even explains how to create own patch nodes. So, this has to be done first. Well, that’s a rational direction for the tutorial evolution.

donovaly, 2019-01-21:
I started with Arduino in general and XOD in special just 24 hours ago and I am on fire. I want to give something back, so if you give me few days I think I will be ready to contribute to the tutorial. How can/should I send proposed tutorial patches?

Many thanks for all your great work!

donovaly, 2019-01-22:
> Feel free to share an updated project as a xodball or even make a pull request.

OK, I started with a xodball. I created #1643 for the patch to separate my patch from this issue.

donovaly, 2019-01-23:
Please also add this link somewhere to a comment:
https://xod.io/docs/reference/shortcuts/
This is very valuable for beginners to get info about shortcuts. I propose in the section about Help.

nkrkv, 2019-04-16:
Closed by the 0.29.0 release

closed 2019-04-16 by nkrkv

donovaly, 2019-05-05:
Great work! Many thanks