xodio/xod #1634: patch: make DHT11 sensors working with XOD

donovaly, 2019-01-21:
### Environment

* Scope: desktop IDE
* Version: 0.26.1
* OS: Windows 7
* Machine: i7, 8 GB RAM

### description

(At first apologize for mistakes, because I just started with XOD.)

I wanted to use a DHT11 sensor with XOD but don't get any output while I get them via an Arduino sketch. I thought about it and found the solution:

Sensors that measure environmental properties need some time to perform the measurement. E.g. to get a humidity value you can rely on the sensor needs to measure it for a certain time before he can deliver a value. Therefore only cannot trigger those sensors "Continuously" but need to specify a time interval. This interval needs to be large enough that a measurement can be performed and output before the next measurement starts.

### How to fix

So the simple and very effective solution is to connect a clock to the UPD input of the existing DHT11 node. Attached is a patch that does this:
[dht-fixed.zip](https://github.com/xodio/xod/files/2779358/dht-fixed.zip)

Please consider my patch because this is a general issue of XOD that one cannot adjust the read out time interval. Other users have the same problem, see e.g. this discussion:
https://forum.xod.io/t/dht11-showing-correct-values-in-the-arduino-ide-but-0-values-in-xod/1096/9

Of course every environmental sensor has other interval limits, therefore the existing DHT11 node works for some DHT devices and fails for others. Therefore when my patch would be applied, set the default interval time to 20 ms. I experimented a bit and found out that my DHT11 needs at least an interval of 10 ms (9 and lover fails) but other devices might need 11 or 12 ms.

### Affected nodes

In my opinion this issue applies not only for the DHT11 node, but I am too new to make a sensible statement.

donovaly, 2019-01-24:
> Can you confirm you get ERR’ors when reading a DHT11 with the xod/common-hardware/dht11-thermometer node and get sane values when reading with Arduino IDE and the same circuit?

I investigated a bit and it seems that my solution is right, but not the 20 ms interval I proposed.

Using the Arduino IDE it works with this code fine:
```
#include <dht_nonblocking.h>
#define DHT_SENSOR_TYPE DHT_TYPE_11
static const int DHT_SENSOR_PIN = 2;
DHT_nonblocking dht_sensor( DHT_SENSOR_PIN, DHT_SENSOR_TYPE );
void setup( ){  Serial.begin( 9600);}
/*
 * Poll for a measurement, keeping the state machine alive.  Returns
 * true if a measurement is available.
 */
static bool measure_environment( float *temperature, float *humidity )
{
  static unsigned long measurement_timestamp = millis( );
  if( dht_sensor.measure( temperature, humidity ) == true )
    {
      measurement_timestamp = millis( );
      return( true );
    }
  return( false );
}
void loop( )
{
  float temperature;
  float humidity;
  /* Measure temperature and humidity.  If the functions returns
     true, then a measurement is available. */
  if( measure_environment( &temperature, &humidity ) == true )
  {
    Serial.print( "T = " );
    Serial.print( temperature, 1 );
    Serial.print( " deg. C, H = " );
    Serial.print( humidity, 1 );
    Serial.println( "%" );
  }
}
```
Interestingly the code does not output faster than about every 2.2 seconds:

[image]

With the same circuit and this XOD program:
[image]

I need at least an IVAL of 0.010 to see anything in the Watch nodes. If IVAL is >= 0.01 I see most of the time ERR and only a few times OK.
First if IVAL is >= 1.2 I see permanently OK.

nkrkv, 2019-01-24:
OK, so to keep the focus, let’s make this issue about throttling updates of DHT11.

### How to fix

- Place a throttle node between `UPD` of `dht11-thermometer` and `dhtxx-read-raw`
- Set throttling interval to a value of 20ms

brusherru, 2019-09-26:
Done within https://github.com/xodio/xod/pull/1853
But set the throttling interval to 200ms, otherwise, it fails sometimes :(

closed 2019-09-26 by brusherru