xodio/xod #1635: enhamcement: add feature to replace nodes

donovaly, 2019-01-21:
### Environment

* Scope: all
* Version: 0.26.1
* OS: all
* Machine: all

### Problem

A common task is to replace nodes by another one. Take for example the mapping node. In many cases you later see that you need to replace it with the map-clip node.

### Solution

add the following feature:
- right click on a node and select in an appearing context menu "replace"
- this will open the search nodes dialog and you can search for a node you like
- after selecting a node and pressing Return the selected node will replace the existing one
- links to the former node will automatically be removed
(- if possible links to inputs/outputs whose name and type is identic to the former node could stay)

nkrkv, 2019-01-23:
Thanks for the feedback. I see the problem, but can’t agree with the solution.

Asking to do a new search for a replacement is excessive for the most common substitutions. It could be done with the drop-down in the Inspector (under node name)

[image]

The list could be populated automatically. But the concrete algorithm should be carefully thought: full match by pins, full match by pin types, strict superset by pins, whatever?

Another way to think about is having an ability to replace an existing node with another _already_ existing node on a patch. Some tricky mechanism to drag-n-drop.

Yet other way is your proposal supported by the new special syntax in the search bar. Something like ”replaces:xod/core/map foo bar baz” which will look for “foo bar baz” but only among nodes suitable to replace `xod/core/map`.

So, the issue poses more questions than aswers and so I can’t accept it to the development backlog. A good idea is to [share the thoughts with the community](https://forum.xod.io/) to collect opinions. Then the issue can be re-opened with all UX aspects detailed.

closed 2019-01-23 by nkrkv

donovaly, 2019-01-23:
> So, the issue poses more questions than answer

Please let's keep it simple! I described the implementation of LabView and it works there very well and saves a lot of time. There is no need to make it that complicated to keep links. Just replace it by the desired node, cutting off/deleting all existing links to the existing/former node is fine.

nkrkv, 2019-01-24:
I’ve moved the discussion to https://forum.xod.io/t/add-feature-to-replace-nodes/1766. There it is more visible to the community.

donovaly, 2019-01-24:
Hmm, I see that your solution leads to what one needs - a replacement to the same position. So a real replacement feature would only make sense if some links would be kept alive. I think that is too much for XOD for now. Thus from my point of view you can close this issue if you like.

p.s. Why can't I close issues I opened when commenting? This works for other github projects?

nkrkv, 2019-01-24:
> p.s. Why can't I close issues I opened when commenting? This works for other github projects?

That’s strange. It should be a button to the left of “Comment”. Do you see it? Or maybe the issue you’re talking about is closed already?

donovaly, 2019-01-24:
> It should be a button to the left of “Comment”. Do you see it?

No. In other GitHub projects I contribute like for OpenSlides I see there a button "Comment and Close".