xodio/xod #1636: missing info about continuously input-pulse

donovaly, 2019-01-21:
* Version: 0.26.1

The 
xod/patch-nodes/input-pulse
offers to input 3 types: Never, On Boot, and Continuously

There is no info in the docs or help what "Continuously" means. Does it mean every millisecond a pulse is triggered?

This info is important because "Continuously" fails as UPD input for some environmental sensors, see issue #1634 as example. therefore this info is important in case you are investigating an error.

nkrkv, 2019-01-23:
Thanks for sharing

The information about “Continuously” is not directly related to `input-pulse` because a xoder will stumble upon it in many other places. It is a higher-level concept which is mentioned (obviously, too briefly) at https://xod.io/docs/reference/data-types/#pulse-literals.

In other words, we should not describe what a pulse is in the short `input-pulse` description because it’s short. Instead, give the knowledge in docs and tutorial. Does it make sense?

Could you think about where and what docs should be improved so that the confusion you’ve experienced is cleared?

nkrkv, 2019-01-23:
#1427 is related

donovaly, 2019-01-24:
> Instead, give the knowledge in docs and tutorial. Does it make sense?

It is valuable OK to describe it in the tutorial. I don't think I get confusions. Please keep the tutorial as it is, but add info:

In the tutorial you write
"* `Continuously`: Generate pulses constantly, i.e. take the readings or make updates with the highest possible rate."

I miss is what "highest possible rate" means. So the info that the rate depends on the board and the connection to it determines the rate would be helpful.
For the docs (maybe even in the tutorial) it would be helpful to add a link how to measure the rate for the own setup.

nkrkv, 2019-02-25:
Superceeded by #1427 and #1631

closed 2019-02-25 by nkrkv