xodio/xod #1637: enhancement: add feature to bend links

donovaly, 2019-01-21:
### Environment

* Scope: all
* Version: 0.26.1
* OS: all
* Machine: all

### Problem

When you have many links you loos the overview quiclkly. Therefore XOD offers buses. However, sometimes XOD should look cleaner also in a patch has only few nodes. Take for example this:

[image]

There it would look cleaner if the links to the watches would have a curve.

### Solution

Add a feature to bend links. in most cases it would be sufficient if links could have one curve. Take as reference the curve line connector in LibreOffice draw:

[image]

nkrkv, 2019-01-23:
This was discussed before and we refuse to accept the feature in this way

> That’s a very questionable thing that did not get a consensus yet. The problem is not in bending and curves per se but is in the fact everyone wants to see a clear graph without intersections, overlaps, etc. If we’d give arbitrary control of link shape I afraid that would be perfectionist hell. Instead of making a program, he would spend most of the time in search for a nonexistent layout perfecto. I’m struggling to find a way when all links would bend automatically to achieve the best readability result.

There’s a `xod/patch-nodes/jumper` which can be used as a dummy to change the visual link path if absolutely required. We should change its face to be rendered specially, as a straight line, without the rectangle and label.

nkrkv, 2019-01-23:
As we have #1645, I’m closing this issue for now.

Also, consider using [buses](https://xod.io/docs/guide/buses/) when you get too many links.

closed 2019-01-23 by nkrkv