xodio/xod #1638: enhancement: allow to start debugger any time

donovaly, 2019-01-21:
* Version: 0.26.1

### Problem

One often has the case to debug running programs. Currently one can upload a program to the Arduino and start a debug session right afterwards. This is a limitation because once a program is upload it is running so when I stop a debug session, I want to restart the debugger (without changing the program) maybe half an hour later.
Currently to restart the debugger I have to upload the program again first - despite I did not change anything. If technically possible, a debug session should be possible to start also without a reupload.

nkrkv, 2019-01-23:
We can’t be sure that the board on the other end of the cable is the same board we used to upload the current version of the current program. So, there’s a challenge. We can try to establish communication and exchange program fingerprints. This involves notable XOD debug protocol update.

The feature makes sense but should be carefully detailed with all edge-cases described. I’m putting it on hold now.

Cesar-S, 2019-01-24:
There is something that also bother me a little, is to have to connect a plate to compile, although not marked debug.

Sometimes I just connect the board to authorize and then disconnect it, a simple Port COMM is enough to start a compilation...

nkrkv, 2019-01-24:
OK, so a yet better solution might be splitting the three functions:

- Compilation
- Upload
- Debug

That is, we could make them available one by one. And have a shortcut (in a general sense) to perform all three at once.