xodio/xod #1640: enhancement: add scrollbar to comment nodes

donovaly, 2019-01-22:
### Environment

* Scope: desktop IDE
* Version: 0.26.1

### Problem

One often needs to write some background info to code but one needs to keep the overview. Therefore comments cannot be as large as their content. Currently XOD truncates the comment content if the comment size is not large enough without any hint for the reader about the truncation. This is a problem.

### Solution

Add a scrollbar to comment nodes. This way users see that there is more in the comment than currently shown and they can either scroll through

nkrkv, 2019-01-23:
Thanks for the feedback. I’ve never had a desire to have comment content bigger that it can contain. I believe it can be handy in some cases though.

The problem is in the solution proposed. It has a serious flaw. The regular scroll bar will make mouse wheel rotation (or two-finger touchpad swipe) ambiguous and modal. Should it pan the patch board or scroll the comment now? Depending on the cursor position we get wildly different effects. Effectively, we create a scrolling container inside a scrolling container. This is a bad UX.

Another idea is required. I’m closing the issue right now, but feel free to re-open it if a non-conflicting idea appears. It is a good idea to run a discussion on the [forum](https://forum.xod.io/) as well.

closed 2019-01-23 by nkrkv

donovaly, 2019-01-23:
> I’ve never had a desire to have comment content bigger that it can contain. 

I am coming from the world of LabView. There you usually have more comment content than actual code. This is normal and my former bosses even pushed us to document each and every bit and now, with some years experience I see how valuable proper documentation is in real life applications.
So if XOD should be an application for real life projects filling comments with lot of info is definitely necessary.

> The regular scroll bar will make mouse wheel rotation (or two-finger touchpad swipe) ambiguous and modal. Should it pan the patch board or scroll the comment now?

It should scroll in the comment when the cursor is inside the comment. If the cursor it outside the comment, the scroll wheel acts like it currently does.

> Effectively, we create a scrolling container inside a scrolling container. This is a bad UX.

I don't agree, see e.g. this Github report window:
I write in a field container inside a scrollable website. So that is exactly the same as I proposed for XOD.

Thus please reopen this bug to be able to reference it in forum discussions and that it won't be forgotten. I really insist that this is an important feature for larger projects.

nkrkv, 2019-01-24:
Moved the discussion to https://forum.xod.io/t/add-scrollbar-to-comment-nodes/1767