xodio/xod #1643: patch: 3 new tutorial sections

donovaly, 2019-01-22:
* Version: 0.26.1

I am a newbie and love XOD. I want to give something back and as mentioned in #1631 I think the tutorial deserves improvements.
Therefore I created 3 new tutorial sections:

[tutorial-props-docs-tests-debug.zip](https://github.com/xodio/xod/files/2781111/tutorial-props-docs-tests-debug.zip)

From my beginner perspective I missed the most
* how to debug/watch
For me I needed it when the servo came into play, then also for the thermometer etc. Thus watching is so important to be able to step through the tutorial that this section should come after "section 10-math". Therefore I used the setup of section 10 for my "xx3-debugging" section.

I also missed
* how to document and how to use comments
This might seem clear but for me it was not. And therefore I think a short section to see what XOD offers to document code doesn't harm. This section can be put anywhere where you think it fits best.

* Finally I created a section to describe the new and very helpful feature of tabtests. This should be the second to last section of the tutorial.

nkrkv, 2019-01-24:
This issue is about new tutorial sections. So, to keep the track I’m copying a comment of @donovaly from #1631 

> Please also add this link somewhere to a comment:
> https://xod.io/docs/reference/shortcuts/
> This is very valuable for beginners to get info about shortcuts. I propose in the section about Help.

donovaly, 2019-01-24:
The reference to the shortcuts should not be in one of the 3 new sections I proposed in this issue but should be added to the existing Help section of the tutorial.

nkrkv, 2019-01-24:
Yes, I agree. I’m collecting the pieces that should be added to the existing material.

nkrkv, 2019-02-25:
Superceeded by #1631

closed 2019-02-25 by nkrkv

donovaly, 2019-02-26:
> Superceeded by #1631

OK, but will the next release have the new tutorial sections? I ask because you said it will be a task for the last release but it came with an unchanged tutorial.

nkrkv, 2019-02-26:
We’re agile and has to sacrifice some scope to keep things running. To be specific, we were running out of time the last release, so we have to choose: whether we want to update the tutorial or implement tweak nodes besides the `tweak-number`. We chose the latest.

Now we are very close to starting the effort of re-writing the tutorial. The work may be started in a few days. However, it is not so obvious should we postpone the whole release for the re-work or it makes more sense to release what we have quickly and then dedicate a whole release to the new tutorial. It’s an org question yet to be solved.

In any case, the new tutorial is a matter of a month or so.

donovaly, 2019-02-26:
> we were running out of time the last release

Hmm, and I thought you are an OpenSource software and can release whenever you like. However, I hope my 3 proposed tutorial sections helped the project somehow.

nkrkv, 2019-02-27:
Open source means the sources are open and the license is free, that’s it. The release schedule is another dimension. There are many chaotic projects in the open source world as well as there’s many well-organized. For example, consider Ubuntu Linux distribution. It’s open source but always released twice per year: in April and October.

What’s for your sections, yes, they _are_ helpful and thank you for them. We’re going to use them (modified or not) within the brand-new tutorial.