xodio/xod #1675: Upload to ESP32 is broken

nkrkv, 2019-02-08:
A report from forum PM. Unconfirmed, but with many useful details

* * *

[...]

I have been looking in to the compile problem with XOD and ESP32/Arduino-cli.exe
i am not well versed in in these things but i noticed that the CLI compiles in to a temp directory
for example: "C:\Users\Brad\AppData\Local\Temp\arduino-sketch-5AA8FAF0366A627054B476A77B6EA40D"

Perfectly fine then the uploader attempts to upload from a different Temp directory
Example:"C:\Users\Brad\AppData\Local\Temp\xod_temp_sketchbook7UeriD\xod_1549527156480_sketch"
it works perfectly except for the fact that only the “sketch.ino.bin” and “sketch.ino.elf” files are copied in to this directory as well as saving the “sketch.ino” there.

I am not sure if this part between compile and upload is handled by XOD or the compiler but it misses the “sketch.ino.partitions.bin” and that appears to be causing the error on upload (if i copy it manually to the XOD temp folder it uploads and runs perfectly…) "[Errno 2] No such file or directory: ‘C:\Users\Brad\AppData\Local\Temp\xod_temp_sketchbook7UeriD\xod_1549527156480_sketch/xod_1549527156480_sketch.esp32.esp32.lolin32.partitions.bin’"

It appears this problem is for ALL ESP32 boards.
[...]

nkrkv, 2019-03-15:
This is a bug in `arduino-cli` https://github.com/arduino/arduino-cli/issues/163

We could copy/rename `*.partitions.bin` from the build directory to the sketch directory (where `arduino-cli upload` expects them) as a temporary workaround

brusherru, 2020-09-23:
Done in #2038

closed 2020-09-23 by brusherru