xodio/xod #1687: Suggestion for pins output-self input-device & output-device

Cesar-S, 2019-02-14:
By double clicking, they are currently deepening and they are pins without content that are shown in the patches.

They could be like the constants that when you click, you go to the inspector's box to change their name.

nkrkv, 2019-02-14:
Sounds reasonable. Thanks

Cesar-S, 2019-02-14:
Thanks for accepting the suggestion.

I thought about the pins in the browser mainly, but I think it should be for all the pins, or I'm jumping something ... what do you think?

nkrkv, 2019-02-15:
> but I think it should be for all the pins

Do you mean that double-clicking a node body (drill-down) should be different from double-clicking its pin (focus its value in Inspector)? If so, that’s a brave idea, yet controversial. I see potential. It’s not directly related to this particular issue and requires discussion. Feel free to open a thread on the [forum](https://forum.xod.io). We use GitHub issues to track tasks, not for discussions.

Cesar-S, 2019-02-15:
Sometimes with some images I can explain better. I do not get along with the translator.
I opened a discussion here https://forum.xod.io/t/double-click-on-pins/1875

brusherru, 2020-11-16:
The recent PR ( https://github.com/xodio/xod/pull/2063 ) is related to this issue and probably closes it until we want to focus on pin values by double-clicking some of the terminals.
If so, please open another issue.

closed 2020-11-16 by brusherru