xodio/xod #1715: Label y value in constants, suggestion ...

Cesar-S, 2019-03-07:
Using the new XOD 0.28.0, a very important change, just a small suggestion.

When you label constants you lose sight of the value.

[image]

nkrkv, 2019-03-11:
Would you elaborate a bit? Are you talking about `MED` is hiding `25` inside it? If, so this is (kinda) by-design. When you give any node a label, that label completely overrides what’s written on the node image. So, you can label your constant `π`, for example, and make it equal to 3.1415926.

Most likely it is a bad design decision and XOD should never completely override the original label. One idea is always leave the original label as is and if custom one is provided, place it at the tail in addition:

- `led: Red`
- `button: Start`
- `3.1415926: π`

It requires #937 though and relates to all nodes, not only the constants.

Am I understood you correctly?

Cesar-S, 2019-03-11:
Hi, something as simple as MED: 25, sorry, had not occurred to me.
Example: I define constants but if I leave numbers I forget they are, and if I add a tag to the naked eye I do not see the values.
It is only for "number" and "byte", but defining a MED: 25 tag is a good solution. Thanks