xodio/xod #1721: Difficulty in selecting and moving buses

Cesar-S, 2019-03-15:
The area to select a `bus` is very small, and most of the time I end up selecting the `pin`. 🤦‍♂️  

Could a greater hidden area be made that occupies at least 50% of the cell? or invert the triangle but maybe it is not right.

[image]

nkrkv, 2019-03-15:
It might be not obvious, but you can also drag by the label.

I agree however, a bus should detect hover whenever the cursor is put anywhere above the slot occupied by it.

Here’s a GIF demonstrating the current state.

[image]

Cesar-S, 2019-03-15:
My problem arises here, I'm going to try to get used to selecting on the label.

[image]

nkrkv, 2019-03-15:
The pins’ hotspot is bigger than the visual circle by-design. Otherwise, it would be too hard to point the pins while linking. We should polish bus node image somehow so that a xoder would not be tempted to drag it by pin.

Cesar-S, 2019-03-15:
If I select the bus or node, the pins are active to connect, the connection could be deactivated when the node is selected.

PR #2069 (2020-11-16): Restyle bus nodes to avoid misclicks
It closes #1721.

Meet the new look of bus nodes:
[image]

Please, check it out, try to drag, select nodes, try to overlap some nodes, and then drag it to other places and etc.
closed 2020-11-16 by brusherru