xodio/xod #1723: Path to arduino-cli config should be enquoted

nkrkv, 2019-03-15:
See: https://forum.xod.io/t/problems-compiling/2020

### Environment

* Scope: desktop IDE
* Version: 0.28.1
* OS: Windows 10

### Steps to reproduce

1.  Start XOD on behalf of a user with spaces in the name (ex: `C:\Users\Foo Bar Qux`)
2.  Hit Deploy → Upload

### Expected behavior

I can choose a board and upload successfully

### Actual behavior

I get an error saying «… unknown command “Bar” for “arduino-cli” … »

### How to fix

Enquote the argument of `--config-file` and other path arguments. Currently, the shell is asked to run (line breaks added for clarity):

```
"C:\Users\Foo Bar Qux\AppData\Local\Programs\XOD IDE\resources\arduino-cli.exe"
--config-file=C:\Users\Foo
Bar
Qux\AppData\Local\Temp\arduino-cli2t6a9o.cli-config.yml
config
dump
```

cologman, 2019-03-15:
Ok, as I am a newby, do I do that in the code I'm trying to compile or in the source code for XOD. If its the later it is well outside my pay grade.

nkrkv, 2019-03-15:
The second case

PR #1906 (2019-12-02): Fix arduino-cli and spaces in username on Windows
It fixes #1723
closed 2019-12-02 by brusherru