xodio/xod #1725: Function delay does not work

Cesar-S, 2019-03-20:
### Environment

* Scope: desktop IDE, browser IDE, CLI
* Version: 0.28.1
* OS: Windows 10 
### Steps to reproduce

A node with a pin number time(T), input(IN) and output(OUT) pin.
Use the following code ...

```
struct State {
};
{{ GENERATED_CODE }}
void evaluate(Context ctx) {
    auto inValue = getValue<input_IN>(ctx);
    auto inValueT = getValue<input_T>(ctx);
    delay (inValueT*1000);
    emitValue<output_OUT>(ctx, inValue);
}
```
[delay-bug.zip](https://github.com/xodio/xod/files/2989332/delay-bug.zip)

### Expected behavior

With a value of 25 in T, the output OUT will show the value of IN after 25 seconds.

### Actual behavior

The output OUT shows the value of IN immediately.

* This worked perfect up to 0.25.3, I could not test the later versions for problems in login from xod.io, I can not download 0.26.0 / 0.26.1 / 0.27.0

evgenykochetkov, 2019-03-21:
Thanks for the detailed report!

I can reproduce this behaviour in simulation mode(the `delay` function does nothing there). On real hardware your code works fine for me.

That being said, **please** avoid using `delay` in your implementations at all costs. It will block execution of all other nodes.

The non-blocking version of your example will look something like this:
```
struct State {
};

{{ GENERATED_CODE }}

void evaluate(Context ctx) {
    auto inValue = getValue<input_IN>(ctx);

    if (isSettingUp()) {
        auto inValueT = getValue<input_T>(ctx);
        setTimeout(ctx, inValueT * 1000);
    }

    if (isTimedOut(ctx)) {
        emitValue<output_OUT>(ctx, inValue);
    }
}
```
(Take a look at https://xod.io/docs/guide/cpp-time/ for a more in-depth explanation)

Or just use a combination of `delay` and `buffer` nodes to achieve the same effect:
[image]

Cesar-S, 2019-03-21:
hi, this library  https://xod.io/libs/gabbapeople/esc/ also has a delay

closed 2019-03-22 by nkrkv

evgenykochetkov, 2019-03-22:
> hi, this library https://xod.io/libs/gabbapeople/esc/ also has a delay

Yes, that five second delay on initialisation is... suboptimal. Nice catch :)

An exception to "avoid delays" rule wold be implementing a communication protocol with some device.
Here is an example: https://github.com/xodio/xod/blob/master/workspace/__lib__/xod/common-hardware/ds18b20-thermometer/patch.cpp
Notice that delay times are microscopic and such precision just could not be achieved with `setTimeout `