xodio/xod #1734: Annoying missclicks on pins while you want to drag the selection

brusherru, 2019-03-29:
### Rationale
To drag the selection User have to click (and hold) mouse button on the Node. But if User clicks on a pin it starts linking mode. Canceling linking deselects all.

The tweak suggestion reported on the forum: https://forum.xod.io/t/select-copy-move-various-node-suggestion/2036

**In short:**
Suggested solution: disable pins on selected nodes (when selected more than one?).
I think about another solution: start linking by the `mouseup` instead of `click`.

We have to discuss how to tweak it. If you have any suggestions or thoughts about usability, welcome to the comments here or in the forum topic.

Cesar-S, 2019-03-29:
Move the selected nodes with the arrows?

> I think about another solution: start linking by the mouseup instead of click.

I do not know how it would be the connection with `mouseup`, I would have to see and try.

Sometimes it happens that just by passing the mouse the connection is made and when clicking a new connection comes out that is not required.
To reproduce the error.

1-several pin-out and watch
2-start the connections.
3-after making some, the connection becomes automatic without clicking.
4-When you click, a new connection is released that is not required.
This happens when only the outline is marked and the center of the pin is dark. Maybe it repeats several times and stops when the mause is placed and connects without clicking.

I will try to achieve some captures, it is something that happens several versions ago

brusherru, 2020-11-12:
Oops. It's a duplicate of https://github.com/xodio/xod/issues/1721

closed 2020-11-12 by brusherru