xodio/xod #1762: As a xoder I don't want to loose my project accidentally pressed Ctrl+R

brusherru, 2019-04-18:
Reported on the [forum](https://forum.xod.io/t/ctrl-r-function/2151).

### Rationale

We don't disable or reassign `CtrlOrCmd+R` hotkey. So it could refresh the renderer or web IDE without any prompting, so User will lose all his changes in the project.

Let's disable the hotkey.

To make possible to refresh the renderer it can contain a "Refresh" button in the main menu. But, actually, this is a very rare case when User needs it (only if something went wrong and we didn't handle some error in the renderer). So I think we can just disable the hotkey.

### Acceptance criteria

- [ ] `CtrlOrCmd+R` hotkey does nothing in desktop IDE
- [ ] `CtrlOrCmd+R` hotkey does nothing in web IDE

knopki, 2019-04-18:
You can't disable `CtrlOrCmd+R` in a web browser environment. Also, I think this is bad idea. 
Maybe attach to `window.onbeforeunload` when something not saved and unattach after save?
```js
window.onbeforeunload = function(e) {  return 'nope'; };
```

brusherru, 2019-04-18:
Oh. It's already on the web IDE.
So it's only about the desktop IDE.

But there are some problems with `onbeforeunload` in electron. Need to inspect it.
Here are a few links:
https://github.com/electron/electron/issues/7977
https://github.com/electron/electron/issues/9966

nkrkv, 2019-04-18:
Can the desktop IDE ask to save/discard/cancel on hitting Reload? No matter if triggered by hotkey or the menu.

brusherru, 2019-04-18:
Just tested:
`window.onbeforeunload` works fine in Renderer Process. So we can ask a confirmation, but we have to handle and keep in memory what we need to do next: close IDE or reload 🤔

brusherru, 2019-04-18:
Probably, we don't need to check User wants to reload or close the window. Just ask confirmation in any case on `beforeunload` if there any changes in the project
I tried to implement it, but it caused that I sometimes IDE reloads/closes faster than saves the project.
We need to go deeper 🙈