xodio/xod #1774: xod packages file gone

Vortetty, 2019-05-03:
### Environment

* Scope: desktop IDE
* Version: 0.29.1
* OS: Windows 10 home
* Machine: i5, 8 GB RAM

### Steps to reproduce

1.  download on non-admin account
2.  go to the tutorial uploading section
3.  start the upload to board after plugging in arduino uno (using redboard but mine is identical to the arduino uno, including the port types, just a different brand)
4.  it needs to download libraries
5.  initiate downloading
6.  program crashes
7.  restart and try again
8.  find out i can't choose my board now see error saying file is missing
9.  C:\Users\<username>\AppData\Local\Programs\XOD IDE\__packages__\extra.txt is missing
10.  go to files and see that the __packages__ folder is gone

### Expected behavior

packages should be downloaded then it should upload to my board

### Actual behavior

i cant select my board because C:\Users\<username>\AppData\Local\Programs\XOD IDE\__packages__\extra.txt is missing

### Console output

Error
ENOENT: no such file or directory, open 'C:\Users\vorte\AppData\Local\Programs\XOD IDE\__packages__\extra.txt'
The error has no formatter, which is a bug. Report the issue to XOD developers.

### How to fix

i have no idea as i downloaded his to avoid code problems i often run into when hand typing things. my only guess would be to provide a way to redownload the packages folder with all available packages in it

Cesar-S, 2019-05-03:
hi, what is the board?

Vortetty, 2019-05-03:
The board would be the redboard: https://www.sparkfun.com/products/13975

It is almost identical the the Arduino uno, the main difference being the hardware, however it is essentially the same as the uno software wise. And functions the exact same way. It also has he same pin names.

Cesar-S, 2019-05-03:
Being that it's compatible, I do not think it's a bug.
In Menu -> Deploy -> Upgrade Arduino Packages & Toolchains

nkrkv, 2019-05-03:
Thank you for reporting. That’s weird. I’m sure the board model has nothing to do with the problem. Redboard is fine. In any case that “ENOENT: no such file or directory ... The error has no formatter, which is a bug. Report the issue to XOD developers.” should be handled properly by XOD.

I wonder what your workspace path is? Would you hit File → Switch Workspace? Try creating a new empty directory, set the workspace path to it and try again?

Vortetty, 2019-05-03:
Thank you very much for the idea, I will try it and see| if that fixes it. It this the proper place to report this or would the xod.io site forums be more suitable?

nkrkv, 2019-05-03:
If some behavior is definitely a bug (crash, much confusion, etc) this GitHub repo is the best place to report because XOD developers use it as a source of truth when picking tasks to work on.

Vortetty, 2019-05-03:
Accidentally closed it, I will add screenshots soon.
Couldn't last night

brusherru, 2019-11-14:
I close the issue because of no updates from the author ( @Vortetty ).
If the issue still actual, please, reopen it and provide fresh information and screenshots.

closed 2019-11-14 by brusherru

Vortetty, 2019-11-14:
@brusherru I see no reason to reopen this as a fresh reinstall recently fixed it.