xodio/xod #1775: Compiler Error When Using uart-usb Node

jmwright, 2019-05-08:
This started as a discussion on the forum [here](https://forum.xod.io/t/compilation-error-sending-text-to-a-pc-from-arduino/2215).

### Environment

* Scope: desktop IDE
* Version: 0.29.1
* OS: Ubuntu 18.04 (also reported on Windows 10 by another user on the forum)
* Machine: i7, 16 GB RAM

### What I'm Trying to Do

I'm trying to create a patch that passes string data over the USB port from an Arduino Uno to a PC via the USB-serial bridge.

### Steps to reproduce

1. Hook an Arduino Uno to any USB port on the PC.
2. Create the following program:
[image]
3. Click either the "Upload to Arduino" or "Upload and Debug" button.

### Expected behavior

I expect compilation to succeed.

### Actual behavior

I get a compilation error.

### Console output

A compilation error is shown in the Compiler pane.
***Compiler Pane Output:***
```
Begin compiling code for the board Arduino/Genuino Uno 📦

Using board 'uno' from platform in folder: /home/jwright/xod/__packages__/packages/arduino/hardware/avr/1.6.21
Using core 'arduino' from platform in folder: /home/jwright/xod/__packages__/packages/arduino/hardware/avr/1.6.21

Detecting libraries used...

/home/jwright/xod/__packages__/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/jwright/xod/__packages__/packages/arduino/hardware/avr/1.6.21/cores/arduino -I/home/jwright/xod/__packages__/packages/arduino/hardware/avr/1.6.21/variants/standard /tmp/arduino-sketch-0EBF573157785F19E8DD4087B4EB4780/sketch/xod_1557350307225_sketch.ino.cpp -o /dev/null

Generating function prototypes...

/home/jwright/xod/__packages__/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/jwright/xod/__packages__/packages/arduino/hardware/avr/1.6.21/cores/arduino -I/home/jwright/xod/__packages__/packages/arduino/hardware/avr/1.6.21/variants/standard /tmp/arduino-sketch-0EBF573157785F19E8DD4087B4EB4780/sketch/xod_1557350307225_sketch.ino.cpp -o /tmp/arduino-sketch-0EBF573157785F19E8DD4087B4EB4780/preproc/ctags_target_for_gcc_minus_e.cpp

/home/jwright/xod/__packages__/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino-sketch-0EBF573157785F19E8DD4087B4EB4780/preproc/ctags_target_for_gcc_minus_e.cpp

Compiling sketch...

/home/jwright/xod/__packages__/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/home/jwright/xod/__packages__/packages/arduino/hardware/avr/1.6.21/cores/arduino -I/home/jwright/xod/__packages__/packages/arduino/hardware/avr/1.6.21/variants/standard /tmp/arduino-sketch-0EBF573157785F19E8DD4087B4EB4780/sketch/xod_1557350307225_sketch.ino.cpp -o /tmp/arduino-sketch-0EBF573157785F19E8DD4087B4EB4780/sketch/xod_1557350307225_sketch.ino.cpp.o

/tmp/xod_temp_sketchbook2poidD/xod_1557350307225_sketch/xod_1557350307225_sketch.ino:1151:5: error: 'Serial_' does not name a type
     Serial_ _serial;
     ^
/tmp/xod_temp_sketchbook2poidD/xod_1557350307225_sketch/xod_1557350307225_sketch.ino:1156:21: error: expected ')' before 'serial'
     UsbUart(Serial_ serial, long baud = 9600)
                     ^

/tmp/xod_temp_sketchbook2poidD/xod_1557350307225_sketch/xod_1557350307225_sketch.ino:1189:5: error: 'Serial_' does not name a type
     Serial_* toUsbSerial() {
     ^
/tmp/xod_temp_sketchbook2poidD/xod_1557350307225_sketch/xod_1557350307225_sketch.ino: In member function 'virtual bool xod::xod__uart__uart_usb::UsbUart::available()':
/tmp/xod_temp_sketchbook2poidD/xod_1557350307225_sketch/xod_1557350307225_sketch.ino:1166:22: error: '_serial' was not declared in this scope
         return (bool)_serial.available();
                      ^
/tmp/xod_temp_sketchbook2poidD/xod_1557350307225_sketch/xod_1557350307225_sketch.ino: In member function 'virtual bool xod::xod__uart__uart_usb::UsbUart::writeByte(uint8_t)':
/tmp/xod_temp_sketchbook2poidD/xod_1557350307225_sketch/xod_1557350307225_sketch.ino:1170:22: error: '_serial' was not declared in this scope
         return (bool)_serial.write(byte);
                      ^
[... 1980 lines cut ...]
            xod__uart__print::evaluate(&ctxObj);

            // mark downstream nodes dirty
        }
    }

    // Clear dirtieness and timeouts for all nodes and pins
    node_0.dirtyFlags = 0;
    node_4.dirtyFlags = 0;
    node_5.dirtyFlags = 0;
    node_6.dirtyFlags = 0;
    node_7.dirtyFlags = 0;
    detail::clearStaleTimeout(&node_6);

    XOD_TRACE_F("Transaction completed, t=");
    XOD_TRACE_LN(millis());
}

} // namespace xod

```

### How to fix

I don't know.

jmwright, 2019-05-13:
This ended up being a misunderstanding on my part on what uart-usb was meant for. This is a quote by @nkrkv from the forum:
> The uart-usb node represents a USB serial on the boards having a dedicated CDC interface (like Leonardo or Micro). For Uno and Nano the interface is shared with UART #0. So, if you would use uart-0 instead of uart-usb, all should work fine for you.

I tried using uart-0 and everything worked as intended.

@nkrkv I see that you've already changed the tag from triage to s:stdlib, so I won't close this issue. Thanks for the help. I marked the forum thread as resolved.

nkrkv, 2019-05-13:
Yes, thank you. I’m pretty sure others can be confused too. The `uart-usb` node should fall back to UART #0 if required, or at least provide a clear error message. Let’s keep the issue open to fix it eventually.