xodio/xod #1778: Implement throttle node

nkrkv, 2019-05-14:
The node is necessary to limit some sensors update rate. Having it implemented as a generic would make it handy in many other use cases.

### Node

Throttles input changes to happen no more often than once per interval specified. Useful to limit the rate of updates.

| Dir | Pin  | Type    | Default | Description
| --- | ---- | ------- | ------- | -----------
| in  | IN | t1    |       | Value or pulse to throttle
| in | T | Number | 0 | IN will propagate to OUT at most once per T seconds
| out | OUT | t1 |   | The throttled output. Follows `IN` immediately if `T` time window has passed since the last update, otherwise conducts `IN` with a lag so that the time span between updates is no less than `T`

### Time diagram

```
in -----x----x-x-x-x-x-x-x----x-------------------x--->

out ----x---------x---------x---------x-----------x--->

        |_________|_________|_________|_________| |___
             T         T         T         T
```

In other systems this is called throttling with leading *and* trailing edge.

### How to test

Tabtest which replicates the diagram shown

Cesar-S, 2019-05-15:
Hi, I have a question, it would be something different from this (I call it pulse-timed), as it would be generic? that's why my question .

[image]

nkrkv, 2019-05-15:
As I can see, the patch you’re showing *is* pulse throttling but it only takes into account the leading edge. Compare:

```
Leading (yours)

in -----x----x-x-x-x-x-x-x----x-------------------x--->

out ----x----------x----------x-------------------x--->

        |_________||_________||_________|         |___
             T          T          T         

Leading + Trailing (proposed)

in -----x----x-x-x-x-x-x-x----x-------------------x--->

out ----x---------x---------x---------x-----------x--->

        |_________|_________|_________|_________| |___
             T         T         T         T
```

It’s very useful too. Each variant would find applications in different cases.

Cesar-S, 2019-06-01:
Try to do the node, I think it works but I can not eliminate the initial pulse.

If I send a pulse and before the end of time send other pulses, it will result in a final pulse.

[image]