xodio/xod #1836: Simulation Not working

sfranzyshen, 2019-08-03:
### Environment

* Scope: desktop IDE
* Version: 0.30.1 
* OS: Windows 10 
* Machine: i5, 12 GB RAM

### Steps to reproduce

1.  From the "Welcome-to-XOD" Selected "002-simulate"
2.  From the "Deploy" Menu selected "Simulate"
3.  The Following Error occurs 

```
Generating C++ code for simulation…
Building…
shared:WARNING: object /ws/arduino_data/EM_CACHE/asmjs/libc.bc is not valid according to llvm-nm, cannot link
undefined:1

SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at eval (eval at globalEval (/ws/arduino_data/packages/xod/tools/emscripten/1.38.30/src/compiler.js:110:8), <anonymous>:266:19)
    at eval (<anonymous>)
    at globalEval (/ws/arduino_data/packages/xod/tools/emscripten/1.38.30/src/compiler.js:110:8)
    at load (/ws/arduino_data/packages/xod/tools/emscripten/1.38.30/src/compiler.js:66:5)
    at Object.<anonymous> (/ws/arduino_data/packages/xod/tools/emscripten/1.38.30/src/compiler.js:185:1)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
shared:ERROR: 'node /ws/arduino_data/packages/xod/tools/emscripten/1.38.30/src/compiler.js /tmp/tmpSBmWUN.txt /ws/arduino_data/packages/xod/tools/emscripten/1.38.30/src/library_pthread_stub.js' failed (1)

The generated C++ code contains errors. It can be due to a bad node implementation
or if your board is not compatible with XOD runtime code. The original compiler error
message is above. Fix C++ errors to continue. If you believe it is a bug, report the
problem to XOD developers.
```

### Expected behavior

Should simulate?

### Actual behavior

Doesn't simulate ... Error

### Console output 
```
internal/process/warning.js:18 (node:100592) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead. writeOut @ internal/process/warning.js:18

xod.io/auth/whoami/:1 Failed to load resource: the server responded with a status of 403 ()

DevTools failed to parse SourceMap: file:///C:/Program%20Files/XOD%20IDE/resources/app.asar/src-babel/bundle.js.map
```

Cesar-S, 2019-08-03:
Hi, this is a cloud problem, no simulation is working today.

sfranzyshen, 2019-08-03:
This is also NOT working from the Web IDE either ....

Cesar-S, 2019-08-03:
No simulation works today, you work in the IDE but it co.pila in the cloud.  You can try a simple patch and it won't work today either.

sfranzyshen, 2019-08-03:
> No simulation works today, you work in the IDE but it co.pila in the cloud. You can try a simple patch and it won't work today either.

is this just happening today? or is this a regular occurrence? can the simulation be done locally? 
how dependable is this environment? I mean the very first thing I try ... doesn't work! Should I bother investing any more time on this project?

Cesar-S, 2019-08-03:
It usually happens, maybe the server is under maintenance.  Previous days ah worked well, it will be normalized soon. Instead of simulating you can use debug with a connected board. 
If node need an external library, the simulator will not work.

sfranzyshen, 2019-08-03:
Can the simulation be done locally? If not ... Can it be added?

Cesar-S, 2019-08-03:
I don't know the plans currently of delopment, local only debug.

knopki, 2019-08-03:
Hi there. In short, it fixed. Fell free to reopen the issue if not.

Really, it's a cloud problem. For some reason, the cloud compiler generates a broken binary for the standard c library and that broken binary populated to the shared cache. So, every cloud compiler worker takes broken binary from the cache and compilation fails.

In the future, we want some self-healing automation for this.

closed 2019-08-03 by knopki

Cesar-S, 2019-09-09:
> Hi there. In short, it fixed. Fell free to reopen the issue if not.
>
> Really, it's a cloud problem. For some reason, the cloud compiler generates a broken binary for the standard c library and that [...]

it passed again, I can't login 🤕

nkrkv, 2019-09-09:
@Cesar-S Fixed. Check it out again, please