xodio/xod #1932: Simulation generates confusion to our xoder

Cesar-S, 2019-12-25:
I have seen and tried to help many, but since nobody reads the previous messages or uses the search, it always repeats itself.  

Because there is an error using libraries or the limit of cloud compilation.  

In my opinion the simulation button should be in another place or show a pop-up explaining the limitations and recommending to use debug.

nkrkv, 2019-12-25:
Thank you for narrowing down a problem. I tend to agree. At the same time, I’m sure that the experience should be as straightforward as it can be in the cases when everything is OK and _can_ be simulated without a problem.

XOD should show clear errors with clear suggestions on how to solve the problem if and only if there’s a problem. Obviously we fail to do it now in some very popular scenarios. Could you suggest **top** 2-3 cases when a new xoder get confusion? Solving these cases alone should bring a lot of improvement.

I’d try to guess:

- A xoder is trying to simulate a patch containing `text-lcd-xxx` node
- A xoder is trying to simulate a patch with a community library which is implemented in XOD
- whatever?

Cesar-S, 2019-12-25:
Sorry, I didn't develop the subject well, it's not my thing :)

- A xoder is trying to simulate a patch containing text-lcd-xxx node
-A xoder is trying to simulate a patch with a community library which is implemented in XOD
-A xoder believes it is the option to load the board
-An xoder is unaware that the la in the debug option can also be simulated.
-An xoder, try to simulate and not login, does not have enough quota.

there may be something else, I don't remember now.

I think most do not read, others may have a language problem.

In my case I often confuse the simulate button when I want to load the plate, because I choose the last one on the right. 🤦‍♂