xodio/xod #1944: Bug, no delete patch.cpp , patch.xodp no full changed.

Cesar-S, 2020-01-24:
### Environment

* Scope: desktop IDE
* Version: 0.32.1
* OS: Windows 10

### Steps to reproduce

Create a nth-input clone node, in main I add nth-input and dataselect.
I copy the content from nth-input to nth-clon, I save.
I add non-implemented-in-xod from dataselect, save.
I delete all content, save.

1.  Create a nth-input clone node, in main I add nth-input and dataselect.
2.  Copy the content from nth-input to nth-clon, save.
3.  Add non-implemented-in-xod from dataselect, save.
4. Delete all content, save.

### Expected behavior

Content removed

### Actual behavior

The content has not been deleted, patch.cpp remains, and code remains in patch.xodp

### ---
video capture [bug.zip](https://github.com/xodio/xod/files/4110508/bug.zip)

I found it when trying to add [non-implemented-in-xod](https://github.com/xodio/xod/pull/1940) on native node

evgenykochetkov, 2020-02-05:
Hi! Thanks for the detailed report (the included video was especially helpful).

The `patch.cpp` (and other attachments) not being deleted was known for quite some time (#1618), but we still did not do anything about it 🙈

The issue with `patch.xodp` looks pretty serious, but I failed to reproduce it (tried several machines, Windows and Mac). You sure it was not an issue with notepad++ not displaying an updated version? Could you please try reproducing it with some other editor?

Cesar-S, 2020-02-05:
Hi, ok, I thought patch.ccp had been resolved.

Regarding patch.xodp I can't reproduce it again.  

Maybe it was something more than affection, the truth that abandons the idea of ​​modifying the nodes by the bad results obtained.

 I close until I find the fault again, thanks for answering the problem.

closed 2020-02-05 by Cesar-S