xodio/xod #1946: Different result, simulation vs debug

Cesar-S, 2020-01-30:
Maybe it's a problem, now I have an alternative solution but I leave the restlessness.

https://forum.xod.io/t/different-result-simulation-vs-debug/3384

nkrkv, 2020-01-30:
Hello and thank you for pointing to the problem with a bug issue. I understand the reason behind the confusion but I see no way to fix it. To be more precise: effort amount to fix it does not match the potential gain.

`atoi` is the standard __C++__ function and it operates on `int` type by default. `int` is two bytes on AVR and four bytes on simulation _and_ 32-bit Cortex-M microcontrollers. In the case reported, it’s up to the author to choose a function other than `atoi` to guarantee the same behavior across all platforms. Hope this makes sense.

For now, I’m closing the issue. Feel free to re-open if new circumstances appear.

closed 2020-01-30 by nkrkv

Cesar-S, 2020-01-30:
I understand, as long as it is not something that affects the simulation in general it makes no sense to devote more effort.