xodio/xod #1955: Automatic library download vs local library

Cesar-S, 2020-02-09:
Automatic library download removes the higher version local library.

If the local version is a higher version, you should get a warning that the library is going to downgrade local library.

I lost library improvements for this reason.

nkrkv, 2020-02-10:
I’m sorry to hear you lost a portion of work. But are you :100: sure that the root cause was an automatic installation of a transitive dependency? By design, if XOD finds a required library installed locally it goes on and does not touch it regardless of the version.

If it is indeed a bug, could you please to narrow-down it to a 1-2-3 scenario that is easily reproducible? See https://github.com/xodio/xod/blob/master/.github/ISSUE_TEMPLATE/bug.md

Cesar-S, 2020-02-10:
Hello, thanks for answering.

Reviewing I see that an error of mine, not something automatic.

It can be a mishandling of my bookstores, I recognize it. I should have a copy external .

I would only like that if I am going to install cesars/utils@0.0.25, the system will notify me that it already exists and the version is cesars/utils@0.0.30 , to be possible.

br

closed 2020-02-10 by Cesar-S

nkrkv, 2020-02-10:
Agree, the process of local lib development should be revised and improved at some point.